firmata/arduino #402: Analog read value from 0-810

iridiu, 2018-08-14:
I am usig Standard Firmata Ethernet on a PLC based on Arduino MEGA connected to a 24v power source. If I read analog values, the maximum value is 810, not 1023 and I suppose there is something that it needs to be changed, in my case

soundanalogous, 2018-08-15:
Do you have a multimeter you can use to check the input voltage to the board? I suspect it's less than 5 volts.

iridiu, 2018-08-15:
I have a multimeter but I can't open the PLC without losing warranty, but I used the "range" node in Node-Red and it seems ok, reading voltage and compared with multimeter, is the same. The maximum value is 821...

Here is the link to the PLC: https://controllino.biz/controllino/mega/

soundanalogous, 2018-08-16:
Check the voltage on the 5v pin and see if it's reading as 5v. You may need to supply voltage to the AREF pin if that is the case.

closed 2019-10-29 by soundanalogous