firmata/arduino #404: C++ client for Firmata Arduino

jecalderon, 2018-09-03:
I am looking for a C++  fermata Code for the Windows 10 client.  I am not sure if I am missing something or just one needs to be proposed.

soundanalogous, 2018-09-03:
The code in this repo can also function as a C++ client. You can find a specific example here: https://github.com/zfields/remote-wiring.

zfields, 2018-09-03:
For a more exact example, check https://github.com/ms-iot/remote-wiring. Windows appears to have stopped maintaining it. The current project is at http://remotewiring.com/, but it is strictly c++ and does not have a working c++/cx example for Windows 10.

jecalderon, 2018-09-13:
Thank you all.. but I was no clear.  What I really need is this C++ code within a solution from VS 2017.  This is will all necessary linkers to any GUI directories like WxWidgets.

zfields, 2018-09-13:
https://github.com/ms-iot/remote-wiring

It is literally a c++/cx project. You can simply clone the repository into your project folder and add it to your project via the VisualStudio GUI. Alternatively you can add it via Nuget packages, but I have no idea what version is being supported. If that doesn't work, then you can simply copy and paste the code into anything you have available.

advra, 2019-12-10:
Is there a particular reason none of these are listed under Firmata Client Libraries for C++ within the readme.md? Is it because they are no longer maintained or the current implementation does not support the latest firmata protocol?

zfields, 2019-12-10:
@advra I'm the author of the libraries mentioned above. They are not listed, because they are incomplete _(at least to my liking)_. To my knowledge, the Windows version of `remote-wiring` is no longer actively maintained, but last I knew it worked as intended.
The Linux version is feature incomplete. It currently only supports a serial connection over a USB cable; it's missing transport layers for Bluetooth and Network.
For those reasons, I have elected to not include them in the list. If you can tell me your specific needs then maybe I can help you get going, or set aside some time to complete the transports you need.

advra, 2019-12-10:
@zfields thank you for the explanation. My project will not require Bluetooth but communication over a local network is considered but not a priority. As long as the linux version works such that a Raspberry pi is the host then I can just use the serial USB and delegate the networking tasks to the pi.

zfields, 2019-12-10:
@advra That's great! You should be able to use https://remotewiring.com _(it redirects to GitHub)_. Be sure to log issues on `remote-wiring` and I'll will try and assist you as much as I can!

advra, 2019-12-11:
@zfields Thanks! I will be sure to open any issues I come across. (: