firmata/arduino #406: Standard Firmata PWM Frequency

Ch3fk0ch88, 2018-10-04:
Hello,
i hope someone can help me.
I controll my Mega 2560 with an Pi and NodeRed.
For that i have to use Standard Firmata at the Mega but i have to change the frequency to 25kHz to controll a fan.  I found a way to use 31.37255 kHz put i dont know how to create a 25kHz Signal.

Can Someone help me please?

Thanks

Ch3fk0ch88, 2018-10-05:
Nobody  who did this?

dtex, 2018-10-05:
Sorry for the slow response. Here’s a link to a Johnny-Five issue where we’ve discussed this. Regardless of wether you are using Johnny-Five, the issue is the same.

Ch3fk0ch88, 2018-10-05:
Hi, thanks.
But u forgot to put in the Link ;)

Ch3fk0ch88, 2018-10-07:
Can someone help me please?

dtex, 2018-10-07:
Doh! Sorry 

https://github.com/rwaldron/johnny-five/issues/1469#issuecomment-395122350

Ch3fk0ch88, 2018-10-09:
Hi,
i read it but not really find a solution who i can make it.

dtex, 2018-10-09:
The article linked below is the important part. It may not be possible to get to 25 kHz. It depends on the pins' base clock speed. 

Just curious, what kind of motor/controller is it that requires this specific PWM frequency?

https://playground.arduino.cc/Code/PwmFrequency

Ch3fk0ch88, 2018-10-09:
PC Fans run at 25kHz.

dtex, 2018-10-09:
So this is a four wire motor with PWM on the 4th? If so, this thread may be what you need:

http://forum.arduino.cc/index.php?topic=18742

Ch3fk0ch88, 2018-10-20:
SO i tested everthing i found but i dont get it work...

dtex, 2018-10-22:
Can you share your modified firmata sketch?

closed 2020-11-24 by soundanalogous