firmata/arduino #415: Adafruit Metro M4 (SAMD51) support

Dorme, 2019-01-23:
Hi guys,

Is it possible to implement support for thr Adafruit Metro M4 board (https://www.adafruit.com/product/3382)?
I'm trying to edit the boards.h file, but I keep on getting the error: "Please edit Boards.h with a hardware abstraction for this board".

``#elif defined(ADAFRUIT_METRO_M4__SAMD51_)

soundanalogous, 2019-01-23:
Can you provide a link to the pins_arduino.h and variant.h files for this board?

Dorme, 2019-01-24:
I was able to solve the error I received:
`#elif defined(_VARIANT_METRO_M4_)`

Link to the pins_arduino.h and variant.h files:
[metro_m4.zip](https://github.com/firmata/arduino/files/2790673/metro_m4.zip)

soundanalogous, 2019-01-25:
Seems there is an incompatibility in the Servo class. This has happened in the past with a few boards. Try commenting out the Servo functionality in StandardFirmata and see if it then compiles. You can also check if that Servo library is up-to-date.

Dorme, 2019-01-25:
I've upgraded the Servo library to the latest version available (1.1.3) but it does not solve the issue.
```
...\Servo.cpp: In function 'void Servo_Handler(timer16_Sequence_t, Tc*, uint8_t, uint8_t)':
...\Servo.cpp:44:60: error: 'volatile struct TC_STATUS_Type::<anonymous>' has no member named 'SYNCBUSY'
 #define WAIT_TC16_REGS_SYNC(x) while(x->COUNT16.STATUS.bit.SYNCBUSY);
```
After commenting out the Servo functionality I'm able to compile StandardFirmata and upload it to the Metro M4.

ManOnPajamas, 2019-10-08:
Hello @Dorme ! 
How did you get it work? Which sections did you comment out?

ManOnPajamas, 2019-10-09:
Nvm. Got it! :)