firmata/arduino #417: Arduino Uno wifi rev2 Support

aldis-ameriks, 2019-03-09:
Hello,
Is there a plan to implement support for Arduino Uno Wifi rev2?

Originally, asked the question in johnny-five repo (https://github.com/rwaldron/johnny-five/issues/1542), where it was mentioned that the question would be better suited here, since comes down to firmata support.

soundanalogous, 2019-03-09:
I don't have plans to add support any time soon, but it would be great if anyone from the community takes a first pass at it.

verrol, 2019-08-18:
I would like to see support for Arudino Uno WiFi Rev2 too, I already have to board. So far, I have been using [gobot](https://gobot.io) with my other Arduino and ESP8266 boards, but there is the issue of no WiFi on the other Arduino boards, and the PWM on ESP8266 is not fully supported.

lemio, 2020-05-09:
Same issue here, since Arduino UNO wifi rev2 is used as Arduino Education and firmata is one of the 'core' Arduino libraries, it would be good to add support... 
```
Multiple libraries were found for "Firmata.h"
In file included from /Applications/Arduino 5.app/Contents/Java/libraries/Firmata/Firmata.h:17:0,
 Used: /Applications/Arduino
                 from /Applications/Arduino 5.app/Contents/Java/libraries/Firmata/Firmata.cpp:18:
/Applications/Arduino 5.app/Contents/Java/libraries/Firmata/Boards.h:856:2: error: #error "Please edit Boards.h with a hardware abstraction for this board"
 #error "Please edit Boards.h with a hardware abstraction for this board"
  ^~~~~
/Applications/Arduino 5.app/Contents/Java/libraries/Firmata/Firmata.cpp:19:10: fatal error: HardwareSerial.h: No such file or directory
 #include "HardwareSerial.h"
          ^~~~~~~~~~~~~~~~~~
```

RyderLogisticApp, 2020-08-10:
Is there any plan supporting Arduino Uno wifi rev2? Looking forward to use this in a project. Thanks

fhainz, 2021-01-10:
It would be great to be able to use firmata with an arduino uno wifi rev2!