firmata/arduino #421: [standard mismatch] "Query Firmware" returns file name WITH extension

martin-eden, 2019-05-08:
Hello, [protocol.md](https://github.com/firmata/protocol/blob/master/protocol.md) says    

>The firmware name to be reported should be exactly the same as the name of the Firmata client file, minus the file extension. So for StandardFirmata.ino, the firmware name is: StandardFirmata.

But indeed file name is passed **with** extension.

I think it may be fixed by adding modifying 
[`extension = strstr(name, ".cpp");`](https://github.com/firmata/arduino/blob/master/Firmata.cpp#L207) 
to 
```
extension = strstr(name, ".cpp");
if (!extension)
  extension = strstr(name, ".ino");
```

Issue present in StandardFirmata from Arduino IDE v1.8.8.

soundanalogous, 2019-05-09:
I suspect this has been broken for at least 10 years.

mexchip, 2019-10-29:
As stated by @soundanalogous (https://github.com/firmata/arduino/pull/433#issuecomment-547060097), removing the .ino extension might break some applications, so I think #433 should be closed and maybe this issue replaced by a new one in the protocol project (https://github.com/firmata/protocol), indicating that the firmware name keeps the .ino extension.

mexchip, 2019-10-29:
What do you think of something like this: https://github.com/mexchip/arduino/blob/add-known-issues/readme.md#Known-issues?

soundanalogous, 2019-10-29:
I will fix this in Firmata 3.0 which will include breaking changes. I don't think this needs to be called out specifically in the readme in the mean time. It's captured as an issue which I will keep open, I think that is sufficient.

mexchip, 2019-10-29:
Ok, I've closed the pull request.