firmata/arduino #429: Firmata to support sim800l module.

JGMAGNO, 2019-10-14:
I have been developing nodejs app for a while, and Ive seen this piece of software that enables you to code arduino in a nodejs platform, sadly, I have to control sim800l module, do you have idea how will I control it via firmata?

soundanalogous, 2019-10-14:
Looks like the sim800l uses a Serial interface so it's already compatible with Firmata. Check the Serial interface examples in firmata.js and johnny-five.js.

soundanalogous, 2019-10-14:
Also if you only need to interface with the sim800l and no other hardware, then you can also simply use [Node SerialPort](https://serialport.io/) and don't need firmata at all.

JGMAGNO, 2019-10-15:
if I have ultrasonic sensor(hc-sr04) and sim800l connected to my arduino, how do I control the two of those using firmata? is that possible?

Another question also, its is possible to upload code to my arduino even though it has a firmata on it?

soundanalogous, 2019-10-15:
> Another question also, its is possible to upload code to my arduino even though it has a firmata on it?

Generally no, but at it's core, Firmata is a library rather than just a sketch (StandardFirmata) and you can do whatever you want with it when used as a library, but most people just upload StandardFirmata which has its limitations.

See the following two firmata.js examples for how to interface with a serial peripheral:

1. [Recommended: Using the UART (HW Serial port)](https://github.com/firmata/firmata.js/blob/master/packages/firmata.js/examples/hw-serial-read-gps.js)
2. [Optional: Using SoftwareSerial](https://github.com/firmata/firmata.js/blob/master/packages/firmata.js/examples/sw-serial-read-gps.js)

Documentation for the sim800l should describe the Serial protocol you will need to implement in order to communicate. The examples I referenced above demonstrate how to read over serial. You need to upload StandardFirmataPlus to your Arduino.

However, using the HC-SR04 is problematic since it is not supported by Firmata, at least not at the same time as the sim800l. You'll need to use another microcontroller to interface with the HC-SR04 that then exposes an I2C interface to your main microcontroller running firmata [as described here](http://johnny-five.io/examples/proximity-hcsr04-i2c/). The alternative is to buy an a range sensor that uses either an analog or I2C interface directly such as the HC-SR08, but those are much more expensive.

An alternative is to look into Raspberry Pi or other boards that can run Node directly.

closed 2019-10-29 by JGMAGNO