firmata/arduino #430: PWM flickers when updating

m-elias, 2019-10-16:
I'm testing out the StandardFirmataPlus Arduino example on an Uno clone with node-red-node-arduino on a Pi Zero W. If I send a single analogWrite value, the LED's output is steady. As soon as I send the same repeat or changing values, the LED flickers. On my oscope I managed to capture a flicker and it looks like the Arduino randomly "forgets" to set the PWM output high for one period. My guess is that the firmata code briefly commands a 0 output before updating it to the correct value and right at that moment, the Arduino's PWM output updates and grabs the 0 for one period.

m-elias, 2019-10-16:
Is it changing pin modes? `    case PIN_MODE_PWM:
      if (IS_PIN_PWM(pin)) {
        pinMode(PIN_TO_PWM(pin), OUTPUT);
        analogWrite(PIN_TO_PWM(pin), 0);
        Firmata.setPinMode(pin, PIN_MODE_PWM);
      }
      break;
    case PIN_MODE_SERVO:`

soundanalogous, 2019-10-16:
That code path should only be called once when you initially set the pin mode. Have you ruled out an error in your Firmata client software?

soundanalogous, 2019-10-16:
Also, which pin is the LED attached to?

m-elias, 2019-10-16:
Arduino pin 6. I'm using the node-Arduino library in Node-Red. I'm not sure how to go about debugging that library. I'm very new to Firmata, my goal is to modify it yet to read OneWire sensors and send the data to the Pi so I might switch to a different Node-Red library like contrib-Johnny-Five or contrib-GPIO as from what I read it has more advanced capabilities than node-Arduino. That being said, I'm also not yet sure how/if I can use softserial or a Leo to output debug messages without interfering with Firmata's IO control, to make sure that it's not changing my pinmode. How do people debug in a Firmata sketch? For my purposes, the flicker is not a deal breaker but I wanted raise awareness in case it is a bug somewhere.

soundanalogous, 2019-10-16:
Based on the behavior you're describing, I suspect node-Arduino may set the pin mode each time an analog write message is sent, which would be a node-Arduino issue rather than a Firmata (StandardFirmataPlus) issue. I'd need to investigate that code and the code running on your Pi to know for sure.

m-elias, 2019-10-16:
I think I've found the correct github page for node-Arduino, I'll post there.
Thanks.

soundanalogous, 2019-10-16:
The issue is [here in node-Arduino](https://github.com/node-red/node-red-nodes/blob/master/hardware/Arduino/35-arduino.js#L165). The pin mode should only be set once, but it's set every time a value is sent.

soundanalogous, 2019-10-16:
To answer your other questions:

> my goal is to modify it yet to read OneWire sensors

I recommend johnny-five.js as it has built in support for OneWire.

> I'm also not yet sure how/if I can use softserial or a Leo to output debug messages without interfering with Firmata's IO control

There is a Firmata debug message called STRING_DATA which sends a String from the Firmata firmware to the client library. Here's an example of a [debug string being sent from StandardFirmata](https://gist.github.com/soundanalogous/e10be1c5c0e4031e5aba0769ce5bc1e8#file-standardfirmatalog-ino-L826-L832) and an example of [receiving a debug message in johnny-five](https://github.com/soundanalogous/firmata-client-examples/blob/master/johnny-five/sensors/onewire-plus-i2c.js#L37-L39).

soundanalogous, 2019-10-16:
Looks like [node-Arduino also supports the STRING_DATA message](https://github.com/node-red/node-red-nodes/blob/master/hardware/Arduino/35-arduino.js#L107-L114).

m-elias, 2019-10-24:
My issue was resolved with an update for the node-red-node-arduino package. Thanks for finding the issue.

closed 2019-10-24 by m-elias