firmata/arduino #432: Modify StandardFirmataPlus example to send analog value even if it's unchanged

m-elias, 2019-10-24:
I have modified the StandardFirmataPlus example for my application to send analog readings only every 5 seconds to lower the amount of serial traffic and cpu processing on my firmata host (node-red-node-arduino) and when my application receives the analog value it saves it to a time series database. So I rely on the sent analog value to trigger the db entry but now I realize that firmata is designed to only send data when values change.

Would I be better off modifying the Arduino firmata library to always send analog values regardless of whether they change every 5 seconds or send them more often (every 1 or 0.5 seconds) to my application and limit my message rate there hoping that the value will have changed at least once in the last 5 seconds?

soundanalogous, 2019-10-24:
One solution that is already available is changing the sampling interval (it's 19ms by default). See documentation [here](https://github.com/firmata/protocol/blob/master/protocol.md#sampling-interval).

Another solution that I've put off for way too long is adding a new ANALOG_READ message to the protocol, then the client application can query an analog value as often as needed (well as long as it's an interval > a couple of milliseconds).

m-elias, 2019-10-25:
I'm using node-red-node-arduino which doesn't appear to support changing the sampling period from the GUI nodes but I'm very new to this. I see that node-red-contrib-gpio can set the sampling period but it doesn't support INPUT_PULLUP mode which node-arduino does. Even if I can set the sampling period, if the value doesn't change, will it still send it?

soundanalogous, 2019-10-25:
StandardFirmataPlus sends the analog value each iteration of the sampling interval regardless of if it has changed. Again, the issue lies in node-red-node-arduino which doesn't dispatch the analog value unless it has changed. You can see this [here](https://github.com/node-red/node-red-nodes/blob/master/hardware/Arduino/35-arduino.js#L85).

m-elias, 2019-10-26:
Thanks for explaining it for me. This is my first venture into firmata and I'm learning a lot.

closed 2019-10-26 by m-elias

m-elias, 2019-10-31:
> StandardFirmataPlus sends the analog value each iteration of the sampling interval regardless of if it has changed. Again, the issue lies in node-red-node-arduino which doesn't dispatch the analog [...]

Does this apply to the digital IOs as well?

soundanalogous, 2019-11-01:
It only applies to analog input.

Something that is hard to grasp is that Firmata does not have the equivalent of Arduino's `analogRead()` and `digitalRead()`, instead Firmata uses a reporting scheme where the user enables reporting of either an individual analog pin value or pin changes within a "digital port" (not to be confused with the microcontroller concept of a "port" -- in Firmata a "port" is a sequence of 8 digital pins in order starting from pin 0). If the user enables reporting for a port, any pin values in that port that have changes since the last iteration of the main loop are sent (that is far too frequent to report all digital pin values regardless of if they have changed). If the user enables reporting for an analog pin, that pin value is reported at the rate specified by the sampling interval (19ms by default), regardless if the pin value had changed.