firmata/arduino #434: Arduino as client

atrent, 2019-11-05:
Hi, do I understand correctly that, with the current Firmata lib for Arduino, it is not possible to use an Arduino as "client"? (e.g., having two Arduinos/ESP8266, one slave of the other)

If so, is there any development effort undergoing?

If I'm wrong (as I hope) can you point me towards the right doc/example?

Thank you

soundanalogous, 2019-11-05:
You will have to write an application for one Arduino, but yes the other can use Firmata since the Arduino Firmata library is written for both cases, however there are no example I'm aware of that use Firmata in this way since most user's motivation with Firmata is to control an Arduino using an application written in a language other than C/C++.

soundanalogous, 2019-11-05:
You can also use the Firmata library for communication without using the StandardFirmata sketch (StandardFirmata is just an example use of the library after all).

atrent, 2019-11-06:
Could this be an example "FirmataBlink"? (sort of)

```
#include <Firmata.h>

void setup() {
    Firmata.setFirmwareVersion(FIRMATA_FIRMWARE_MAJOR_VERSION, FIRMATA_FIRMWARE_MINOR_VERSION);
    //Firmata.attach(DIGITAL_MESSAGE, digitalWriteCallback);
    //Firmata.attach(SET_PIN_MODE, setPinModeCallback);
    Firmata.begin(57600); // TODO cambiare seriale
}

void loop() {
    // FirmataBlink (D13 is on port 1?)
    Firmata.sendDigitalPort(1,255); // HIGH (all pins on port 1)
    sleep(1000);
    Firmata.sendDigitalPort(1,0); // LOW (all pins on port 1)
    sleep(1000);
}
```

This would be the client code (on the "master" Arduino) while the other Arduino would be flashed with a StandardFirmata.

zfields, 2019-11-06:
That's not unreasonable, but your implementation uses the legacy `Firmata.h` header and implementation. A while back, I wrote the Firmata Marshaller and Parser to give more fine grain control for client libraries, in fact, it is the implementation invoked by `Firmata.h`

What you are attempting to do, can be made much more simple by utilizing the `FirmataMarshaller` directly.

```c++
#include "FirmataMarshaller.h"

void setup() {
  Serial.begin(57600);
  FirmataMarshaller.begin(Serial);
  FirmataMarshaller.sendPinMode(LED_BUILTIN, OUTPUT);
}

void loop () {
  FirmataMarshaller.sendDigital(LED_BUILTIN, HIGH);
  sleep(1000);
  FirmataMarshaller.sendDigital(LED_BUILTIN, LOW);
  sleep(1000);
}
```

atrent, 2019-11-06:
Thanks! I'll try it.
And you're still implying that on the slave Arduino a StandardFirmata sketch is running?

zfields, 2019-11-06:
Yes, that is correct. Just to clarify terms, we consider what you're calling "slave" as the `host`, and "master" as a `client` of the `host`.

atrent, 2019-11-06:
Yes I am calling:
- master/client an Arduino **instead** of a standard use case PC/Raspberry/etc.
- slave/server/host another Arduino as in the standard use case

umar14, 2019-11-14:
@zfields 

> ```c++
> #include <FirmataMarshaller.h>
> 
> FirmataMarshaller marshaller;
> 
> void setup() {
>   Serial.begin(57600);
>   marshaller.begin(Serial);
>   marshaller.sendPinMode(LED_BUILTIN, OUTPUT);
> }
> 
> void loop () {
>   marshaller.sendDigital(LED_BUILTIN, HIGH);
>   sleep(1000);
>   marshaller.sendDigital(LED_BUILTIN, LOW);
>   sleep(1000);
> }
> ```

Hello, I am trying to accomplish a similar thing. I tried compiling your code but it gives error. 
`'FirmataMarshaller' does not name a type`

I am using the built-in firmata library in arduino ide (v2.5.8). Do I need to change something?

atrent, 2019-11-14:
I think that was sample code to give an idea, the real code should be something like:
```
#include <FirmataMarshaller.h>

firmata::FirmataMarshaller marshaller;

void setup() {
    Serial.begin(57600);
    marshaller.begin(Serial);
    marshaller.sendPinMode(LED_BUILTIN, OUTPUT);
}

void loop () {
    marshaller.sendDigital(LED_BUILTIN, HIGH);
    delay(1000);
    marshaller.sendDigital(LED_BUILTIN, LOW);
    delay(1000);
}
```

umar14, 2019-11-14:
@atrent Thank you so much for the quick reply. I am trying to control arduino mega using esp8266 board and mqtt over WiFi. This just made everything a piece of cake! :grinning:

atrent, 2019-11-14:
I still have to try a setup using ESP8266 with an Arduino slave, did you already try? Did you succeed? Can you post here your setup and final code? Thanks

umar14, 2019-11-14:
Yes I did. Tried it just now and it works flawlessly. I am using NodeMCU board as firmata master/client and flashed the above code you just mentioned. However, one slight change. When flashing to NodeMCU or any other esp8266 board, you would need to change the BUILTIN_LED variable to the actual pin number of the slave (Arduino uno in my case). You specifically have to mention pin 13 and not write BUILTIN_IN.  I flashed the FirmataStandard.ino (present in Arduino IDE) file on Arduino slave. No changes needed. Connected NodeMCU TX pin to Arduino Mega RX pin and GND to GND. That's all. The LED on the slave started blinking. Thank you!

atrent, 2019-11-14:
Yep, the LED_BUILTIN problem (not always mapped correctly on ESP8266 boards) is known. Thanks again

zfields, 2019-11-14:
@atrent Can this issue be considered resolved?

atrent, 2019-11-14:
I think it would be great to include an example in the lib package, then resolved ;)

umar14, 2019-11-14:
An example with callbacks please...

zfields, 2019-11-14:
Loud and clear. The best "example" is probably `Firmata.h`, but I know what you guys are asking for and I'll see if I can't whip something up.

atrent, 2019-11-14:
Yep, an example in the "examples" directory, such as a "RemoteBlink.ino" sketch (to be coupled to StandardFirmata...)
Thanks!