firmata/arduino #435: firmata via xbee on teensy 3.2

nburge, 2019-11-18:
I have been trying to use Firmata on a teensy 3.2, connecting wirelessly by using a pair of S1 xbees. One as a dongle coordinator and the other as a node plugged into the Sparkfun teensy  xbee adapter: https://www.sparkfun.com/products/13311. My aim is to use Pduino (like Maxuino but for Pure Data) to receive three sensor analog values and control some PWM outputs on the teensy.  
Firmata loads and works fine with a USB connection but not via wireless. I'm sure the xbees are connecting, but xbee node to teensy is not.  Is firmata compatible with Soft Serial or AltSoftSerial libraries?

soundanalogous, 2019-11-19:
It can be done with hardware serial. Here's an article I wrote a few years ago that explains how to set it up: http://www.jeffhoefs.com/2013/06/wireless-firmata-using-xbees/.

nburge, 2019-11-19:
Thank you Jeff, yes I've been following the article to set up the xbees. Very helpful. I soldered the pins DIO3 to RTS as described on my xbee explorer board and also on the xbee teensy adapter. When powered up the orange LED flashes twice, short pause then another four times in quick succession. So I think firmata is loading properly. Over USB I can connect from Pduino, flash the orange led, get the version info, send and receive commands.
When I start the Pduino program over a wireless connection via my xbee dongle the red and green Digital In and Out LEDS on the adapter now light up solidly. Commands from Pduino have no effect. 
I understand from the boards.h file that there are three TX RX" pin pairs" on the teensy 3.2. so Serial1 uses pins 0 & 1, Serial2 pins 9 &10 and Serial3 pins 7 & 8
Which Serial port should be used? Do I need to enable more than one serial port in the setup portion of my code? 
(I'm using Standard Firmata.ino.)

soundanalogous, 2019-11-21:
You can use Serial1. See [this section](https://github.com/firmata/arduino/blob/master/examples/StandardFirmata/StandardFirmata.ino#L768-L777) in StandardFirmata.ino. Also comment out line 774 and change line 775 to `while (!Serial1) {`

nburge, 2019-11-21:
IT WORKS!!
A bit of googling also led me to do this....

// Ignore TX1 and RX1 pins
  Firmata.setPinMode(0, PIN_MODE_IGNORE);
  Firmata.setPinMode(1, PIN_MODE_IGNORE);

Otherwise the Tx Rx pins 0 and 1 seem to be set permanently high and cannot function properly. 
Many thanks for your assistance.

closed 2019-11-21 by nburge

nburge, 2019-11-21:
I should add that now there is no flashing sequence of the orange led on pin 13 when booting up. Strange, but not of any importance I suppose.