firmata/arduino #436: Error compiling Firmata for board Arduino Nano Every

eMarek, 2019-12-03:
I would like to install StandardFirmataPlus firmware on my Arduino Nano Every (ATMEGA4809) but it fails with some errors which are unknown to me. If I click "Verify" or "Upload" I get this error:

```
Multiple libraries were found for "Firmata.h"
 Used: /Applications/Arduino.app/Contents/Java/libraries/Firmata
In file included from /Applications/Arduino.app/Contents/Java/libraries/Firmata/Firmata.h:17:0,
Multiple libraries were found for "Servo.h"
 Used: /Applications/Arduino.app/Contents/Java/libraries/Servo
                 from /Applications/Arduino.app/Contents/Java/libraries/Firmata/Firmata.cpp:18:
Multiple libraries were found for "Wire.h"
/Applications/Arduino.app/Contents/Java/libraries/Firmata/Boards.h:856:2: error: #error "Please edit Boards.h with a hardware abstraction for this board"
 Used: /Users/my-username/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/libraries/Wire
 #error "Please edit Boards.h with a hardware abstraction for this board"
  ^~~~~
/Applications/Arduino.app/Contents/Java/libraries/Firmata/Firmata.cpp:19:10: fatal error: HardwareSerial.h: No such file or directory
 #include "HardwareSerial.h"
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board Arduino Nano Every.
```

I am on macOS Catalina Version 10.15.1.
I am using Arduino IDE 1.8.10.
In Library Manager is installed Firmata Built-In by Firmata Developers Version 2.5.8

Any help would be appreciated.

soundanalogous, 2019-12-04:
This is because Arduino Nano Every is not yet supported by Firmata. It needs an entry in Boards.h

Paste the following code into Boards.h:

```c
elif defined(AVR_NANO_EVERY)
#define TOTAL_ANALOG_PINS       8
#define TOTAL_PINS              24 // 14 digital + 8 analog + 2 i2c
#define IS_PIN_DIGITAL(p)       ((p) >= 2 && (p) <= 21) // TBD if pins 0 and 1 are usable
#define IS_PIN_ANALOG(p)        ((p) >= 14 && (p) < 14 + TOTAL_ANALOG_PINS)
#define IS_PIN_PWM(p)           digitalPinHasPWM(p)
#define IS_PIN_SERVO(p)         (IS_PIN_DIGITAL(p) && (p) < MAX_SERVOS) // deprecated since v2.4
#define IS_PIN_I2C(p)           ((p) == PIN_WIRE_SDA || (p) == PIN_WIRE_SCL) // SDA = 22, SCL = 23
#define IS_PIN_SPI(p)           ((p) == SS || (p) == MOSI || (p) == MISO || (p) == SCK)
#define PIN_TO_DIGITAL(p)       (p)
#define PIN_TO_ANALOG(p)        ((p) - 14)
#define PIN_TO_PWM(p)           PIN_TO_DIGITAL(p)
#define PIN_TO_SERVO(p)         (p) // deprecated since v2.4
```

eMarek, 2019-12-04:
Oh, I was not aware that Arduino Nano Every is not yet supported. Is there any roadmap when will it be?

I've tried your code, but I receive pretty much the same error.

soundanalogous, 2019-12-05:
Try changing `elif defined(AVR_NANO_EVERY)` to `elif defined(ARDUINO_NANO_EVERY)`.

eMarek, 2019-12-05:
Tried yesterday already. Didn't work.

eMarek, 2019-12-05:
Today I received Arduino Nano and Arduino Micro where StandardFirmataPlus works so this issue is not on my radar any more. However thanks for the help with Arduino Nano Every.

closed 2019-12-05 by eMarek

jjeff, 2020-11-03:
One year later, this still seems to be a problem.

soundanalogous, 2020-11-07:
https://github.com/firmata/arduino/pull/464

bilal697, 2021-09-12:
#error "Please edit Boards.h with a hardware abstraction for this board"

Can anyone please solve this issue for Arduino Nano 33 BLE Sense? Highly appreciated

pgrawehr, 2021-09-12:
> #error "Please edit Boards.h with a hardware abstraction for this board"
> 
> Can anyone please solve this issue for Arduino Nano 33 BLE Sense? Highly appreciated

Please create a new issue for this, preferably in the firmata/ConfigurableFirmata repo. It will get more attention there.

ThorPrez, 2022-05-14:
Now it´s 2022 and there´s still a problem with Firmata. I got my Nano every today and have changed the boards.h.
I´m trying it with:
elif defined(AVR_NANO_EVERY) || elif defined(ARDUINO_NANO_EVERY)
and get the fault:
In file included from C:\Program Files (x86)\Arduino\libraries\Firmata/Firmata.h:17:0,
                 from C:\Users\ARBEIT~1\AppData\Local\Temp\arduino_modified_sketch_434776\StandardFirmata.ino:28:
C:\Program Files (x86)\Arduino\libraries\Firmata/Boards.h:870:2: error: #error "Please edit Boards.h with a hardware abstraction for this board"
 #error "Please edit Boards.h with a hardware abstraction for this board"

Is there a working solution ?

michaelzoidl, 2022-07-14:
Hey, i just found this issue since I've the same issue for my new _Arduino Nano Every Board_. 
The solution from @soundanalogous is almost right for my board. 

The board identifier for the nano is not `AVR_NANO_EVERY` or `ARDUINO_NANO_EVERY` its [`ARDUINO_AVR_NANO_EVERY`](https://arduino.stackexchange.com/a/85671)

So after pasting this snippet in `~/Arduino/libraries/Firmata/Boards.h` on `L868` (within the if-statement) everything worked.  🎉 
```h
#elif defined(ARDUINO_AVR_NANO_EVERY)
#define TOTAL_ANALOG_PINS       8
#define TOTAL_PINS              24 // 14 digital + 8 analog + 2 i2c
#define IS_PIN_DIGITAL(p)       ((p) >= 2 && (p) <= 21) // TBD if pins 0 and 1 are usable
#define IS_PIN_ANALOG(p)        ((p) >= 14 && (p) < 14 + TOTAL_ANALOG_PINS)
#define IS_PIN_PWM(p)           digitalPinHasPWM(p)
#define IS_PIN_SERVO(p)         (IS_PIN_DIGITAL(p) && (p) < MAX_SERVOS) // deprecated since v2.4
#define IS_PIN_I2C(p)           ((p) == PIN_WIRE_SDA || (p) == PIN_WIRE_SCL) // SDA = 22, SCL = 23
#define IS_PIN_SPI(p)           ((p) == SS || (p) == MOSI || (p) == MISO || (p) == SCK)
#define PIN_TO_DIGITAL(p)       (p)
#define PIN_TO_ANALOG(p)        ((p) - 14)
#define PIN_TO_PWM(p)           PIN_TO_DIGITAL(p)
#define PIN_TO_SERVO(p)         (p) // deprecated since v2.4
```

[Here is my Boards.h file for everyone who wants to see the exact implementation.](https://gist.github.com/michaelzoidl/f398962d63fa54a1200b16441ef81f3c)