firmata/arduino #439: configurable firmata and i2c interfacing (BMP280 --->Arduino Uno ---> PC)

shroffsp, 2019-12-17:
I am wanting to interface a BH1750 --> Arduino Uno --> PC based application using firmata protocol. I have been able to test Arduino Uno  --> PC using a standard firmata and firmata_test.exe application. I have also been able individually interface the BH1750 and Arudino Uno also. I am new to programming and hence not able to figure out how to interface BH1750 along with Configurable Firmata code (I2c case). Can somebody help me in how to integrate the BH1750 code into Configurable Firmata code? So that I can send the read firmata string from the PC application and read the values of the light intensity provided by BH1750. 
What modifications will I need to make to PC firmata string for it? Can Pc based firmata_test.exe app work for I2C case? Is there any such utility to test I2C over firmata available? 
Desperately Looking for help. Any pointers in any directions is welcomed. Thanking you in anticipation

soundanalogous, 2019-12-18:
This sensor uses the I2C protocol so there is no need to add any new code on the firmata side. You can use this sensor with any Firmata client library that supports i2c.

shroffsp, 2019-12-18:
Do I not need to embed the BH1750 code / library along with the Standardfirmata code? As I am new to Firmata, I am not sure how can one call this library in the I2c firmata section? I am seeking help or pointing directions in that area

soundanalogous, 2019-12-19:
I'd need to know more about the application you are trying to develop to be of more help. Are you trying to interface with an Arduino via Firmata via a desktop app? Which programming language are you trying to use. There is an existing solution if you can use JavaScript: See [this example using the BH1750 in johnny-five.js](http://johnny-five.io/examples/light-ambient-BH1750/). No changes are required to StandardFirmata, simply upload StandardFirmata to your board and run that johnny-five script. You can see how this is implemented [here](https://github.com/rwaldron/johnny-five/blob/master/lib/light.js#L389-L426).

shroffsp, 2019-12-20:
I am trying build a LabVIEW based PC application. I want to use firmata (I2Cfirmata) to connect this with a Arduino UNO/Mega board. I want to interface I2c based Sensors like BH1750 / BMP280 etc.. to this Arduino and collect sensor readings from them and provide it to labview based application using Firmata. Based on the sensor readings I want to carry out some control activities in the LabView application.

I am confused how would the I2cfirmata code burnt into arduino directly read the sensors and provide me the sensor reading?  I would need to send some command (Firmata string) to initiate the communication from the LabView end. What command string should I send? 

PS: I donot have any background in javascript programming.

soundanalogous, 2019-12-21:
You would need a LabVIEW Firmata client then. However i'm not aware of one. There are other protocols for interfacing with an Arduino via LabView though. This one looks like it may be helpful for what you're trying to accomplish: https://www.labviewmakerhub.com/doku.php?id=libraries:linx:start.

closed 2020-11-24 by soundanalogous