firmata/arduino #441: Merge Configurable Firmata & Arduino repo ?

Burnich, 2020-01-04:
Hi,

We start using Firmata for some interactive projects mixing computer for UI and Firmata to control HW component.

Some of the HW (like Neopixels) have proprietary protocols, so we need to use the configurable version of firmata to add external library.

The problem is that some change made on the Arduino repository (like the last merge for STM32 support) is not replicate on the configurable version.

I don't really understand why those repo are not merge in one ? The arduino version use mainly the same structure & the only difference is the exemple folder...

If you really need to maintain two repo can we image to move the core files on another repo (Firmata Core) and include this one as a submodule of ConfigurableFirmata & Arduino ?

soundanalogous, 2020-01-04:
If you look more closely you will see this is no simple task. At one point Configurable Firmata was a branch of the Arduino repo, but it became too difficult to maintain as those two branches grew apart. Today the differences are vast. I estimate it may take hundreds of hours to put them back together, especially since the Arduino repo has evolved considerably over the past 3 years.

soundanalogous, 2020-01-04:
The easiest quick fix is to somehow share the Boards.h file across both projects.

Burnich, 2020-01-06:
We are agree on the fact that it will not be straight forward, but I really thing that having those 2 versions separated make firmata adoption & improvement really slower.

If the arduino repository is the most evolving, another proposal can be to add the "Firmata Wrapper Libraries" features to this one and drop the "Configurable" version ?

soundanalogous, 2020-01-07:
The only thing I have time to do is sync the Boards.h files across both repos. I'd need a contributor or two to work on merging the repos.

Burnich, 2020-01-07:
I propably forgot to explain that I can work on it, but before starting I want the community advices & opinions on it !

This issue is a question/debat on the best direction to take before starting making pull request on the repository.

soundanalogous, 2020-01-08:
The Firmata community has been pretty silent the past couple of years. A few years back it was very lively. I'm not sure we'll get to many opinions now though. I suspect a lot of people have moved over to using Raspberry Pi's where you don't need anything like Firmata for the case of people who just don't want to write programs in C.