firmata/arduino #442: johnny-five + firmata on ESP8266, servo not working

benolayinka, 2020-01-15:
I'm trying to control a servo and an ESC on a Wemos D1 Mini using Johnny-Five and StandardFirmata / StandardFirmataWifi. Servo control doesn't work. I confirmed via DMM that on the same pins, writing digital high and low does work, so I know my pin mapping is correct, but I can't get it to drive a servo. LED blink works on pin 16.

I'm running StandardFirmataWifi (and also tested StandardFirmata via usb), so nothing much to show Arduino side, but here's my Johnny-Five code if it helps.

`var five = require("./lib/johnny-five.js");
const {
  EtherPortClient
} = require('etherport-client');
const board = new five.Board({
  port: new EtherPortClient({
    host: '192.168.0.186',
    port: 3030
  }),
});

const PIN_STEERING_SERVO=4 //d2 on esp
const PIN_SPEED_SERVO=14 //d5 on esp
const PIN_LED 16

board.on("ready", function() {
  console.log("Ready event. Repl instance auto-initialized!");

  led = new five.Led(PIN_LED)

  esc = new five.ESC({
          device: "FORWARD_REVERSE",   
          pin: PIN_SPEED_SERVO, 
        });

  servoSteering = new five.Servo({
            pin: PIN_STEERING_SERVO,
            pwmRange: [1000, 2000],
            center: true,
            //debug:true,
          });

  //just testing.. setting values via REPL also does nothing.
  servoSteering.to(120)
  esc.throttle(2000)

  //led blinks, so I know I'm communicating with the board correctly
  led.blink()

  this.repl.inject({
    // Allow limited on/off control access to the
    // Led instance from the REPL.
    servoSteering,
    esc,
    led
  });

});`

dtex, 2020-01-15:
What versions of J5, firmata.js and standardFirmata? Everything needs to be pretty up-to-date to handle the 10-bit PWM.

Also, is it easy to try ```led.pulse()```? If so, do you get full brightness (vs ```led.on()```)?

benolayinka, 2020-01-16:
Firmata 2.5.8 according to library manager, J5 1.4.0 according to package-lock.json.
The same Firmata script and J5 app DO work to control a servo on an Arduino Uno, so I think everything is up to date.
I started getting a J5 'resource is busy' error which I couldn't solve so I replaced the ESP8266 with an Arduino Nano for this project and it's working.
I'll test led.pulse() ASAP and let you know.

benolayinka, 2020-01-27:
> What versions of J5, firmata.js and standardFirmata? Everything needs to be pretty up-to-date to handle the 10-bit PWM.
> 
> Also, is it easy to try `led.pulse()`? If so, do you get full brightness (vs `led.on()`)?

Ok, sorry for the delay. led.pulse() works on GPIO16 and GPIO2 and gives full brightness compared to led.off() (led's are inverted on the esp, so off is on)

benolayinka, 2020-02-01:
> What versions of J5, firmata.js and standardFirmata? Everything needs to be pretty up-to-date to handle the 10-bit PWM.
> 
> Also, is it easy to try `led.pulse()`? If so, do you get full brightness (vs `led.on()`)?

Any idea what else I should try?

closed 2020-09-23 by benolayinka