firmata/arduino #444: I2C read to unexisting address returns a reply

modlfo, 2020-01-27:
Hello,

I'm the developer of the Firmata bindings for Modelica (https://www.wolfram.com/system-modeler/libraries/model-plug/) but I'm facing some issues to make the library more user friendly.

While testing the I2C support, I made the silly mistake of typing incorrectly the I2C device address. It took me a long time to figure out because sending a read request to the incorrect address returns a reply, but there's no information in the reply that notifies me that the device does not exists.

It would be nice having a way of querying if a device with a given address exists, either by having a command that checks if the device can be reached, or including in the I2C reply if the data returned is dummy data (because the device does not exist).

Knowing if the device with the given address exists would make the error reporting much easier.

pgrawehr, 2020-04-05:
I just had the same problem. I found one indication of this, though: If you try to read, say, 1 byte from a nonexistent device, you will get a STRING_MESSAGE reply saying that there wasn't enough data to be read from the bus.

soundanalogous, 2020-04-08:
If you have an idea for a solution, I'll take a pull request. I unfortunately don't have time to work on Firmata anymore so it's going to take community contributions to get things done.

pgrawehr, 2020-04-08:
My suggestion above was just from the client perspective. In Firmata, maybe the bug is that it always returns the expected number of bytes (as zeros), even if there wasn't enough data read from the bus. The bug is probably in https://github.com/firmata/ConfigurableFirmata/blob/master/src/I2CFirmata.h, Line 113. Even though there's not enough data received, the return buffer gets sent with the original number of bytes expected. 

I'm new to firmata/arduino, so I haven't found out how to edit and test a library, but I'll google that later.

closed 2020-04-18 by soundanalogous