firmata/arduino #449: Looking for a new maintainer for Firmata.

soundanalogous, 2020-04-08:
After maintaining Firmata for the past several years, it's time for me to pass on the reigns. My interest in embedded systems & physical computing is just not what it used to be, and new responsibilities in life have made it harder for me to dedicate time to this project. There are [a few different repos](https://github.com/firmata) to maintain. I could pass on some or all. I also have firmata.org registered and would like to pass that on to a new owner. If you are interested, please contact me at the email address listed on my profile page. I want to ensure Firmata is transferred to someone who can move the project forward, but also respect the many projects that depend on it.

soundanalogous, 2020-04-08:
More specifically, I'm looking for someone (or a group of people) who will:

- Review issues and pull requests for firmata/arduino, firmata/protocol, [firmata/ConfigurableFirmata](https://github.com/firmata/ConfigurableFirmata)

dtex, 2020-04-08:
Jeff, I'm happy to help but let's wait a bit and see if someone who is a better fit is interested. If I do it, I'd definitely want to include configurableFirmata and firmatabuilder.com.

Also, a warning: One of the first things I'd do is install stalebot to close issues that have not had any activity in over a year. It will warn everyone on the ticket that it's about to be closed (in a nice way), but if they don't respond in a week it will be closed. They can be reopened of course, but the goal is to cull tickets that nobody still worries about. We did this with Johnny-Five and had no problems. If you hate that idea, then you probably don't want to pass the reins to me.

zfields, 2020-04-08:
@dtex my email is zachary_fields@yahoo.com, if you will send me your email address, I will add you to the Firmata Slack. It may be easier to coordinate this transition there.

soundanalogous, 2020-04-09:
I've still never used Slack for anything other than Firmata... will see if I can still login :D

sidwarkd, 2020-04-09:
I posted a link to this issue on Twitter and somebody suggested Adoptoposs. Not sure if you've heard of them or looked into it.  https://adoptoposs.org/

wallarug, 2020-05-31:
Hey Firmata Team,

Is there a new maintainer for this project yet?

d-a-v, 2020-06-01:
I wonder too how are things doing on Firmata slack !

dtex, 2020-06-02:
Well, we managed to gather a few interested parties in Slack but I don't think @soundanalogous made it in. Should I propose in there that we leave that forum and move it all back out here?

zfields, 2020-06-08:
FYI, @soundanalogous did make it to the Slack channel. However, I think we can increase our transparency and move the discussion out into the public forum.

d-a-v, 2020-06-17:
Did you come to a conclusion ?

pgrawehr, 2020-11-14:
@soundanalogous : As mentioned in https://github.com/firmata/ConfigurableFirmata/pull/96, I'm interested in taking over the responsibility for this project, bringing it forward. I had a talk with @Ellerbach from the https://github.com/dotnet/iot core team, who is interested and ready to help as well. He's been working a lot with microcontroller boards and knows many pieces of hardware and their possibilities. 

If you wish, we can take over the responsibility for the domains and websites as well. I'll send you an E-Mail so we can settle the details, if you're interested.

sheinbergon, 2021-03-27:
So... has a new maintainer been elected?

pgrawehr, 2021-03-27:
@sheinbergon Unfortunately not. After the last round of E-Mail and Slack discussions in November, the communication kind of died off. I had enough of other stuff to do, but I'm still interested.