firmata/arduino #457: Issue when compiling StandardFirmataWiFi for MKR WiFi 1010 board

fredbouchard, 2020-07-11:
## Version
**Firmata:** latest from `master`
**Arduino IDE:** `1.8.13`
**Board:** `Arduino MKRWIFI1010 board` 
**OS version:** macOS Catalina `10.15.4`

## What am I trying to achieve?

I'm trying to compile the `StandardFirmataWiFi.ino` using the `Wifi_NINA` library for the MKR 1010 board.

## Issue

When compiling I'm getting the following error.

```cpp
/Users/{Redacted}/Documents/Arduino/libraries/Firmata/utility/SerialFirmata.cpp: In member function 'void SerialFirmata::checkSerial()':
/Users/{Redacted}/Documents/Arduino/libraries/Firmata/utility/SerialFirmata.cpp:364:56: error: 'SERIAL_RX_BUFFER_SIZE' was not declared in this scope
           if (!((bytesToRead <= 0 && bytesAvailable >= SERIAL_RX_BUFFER_SIZE/2)
                                                        ^~~~~~~~~~~~~~~~~~~~~
/Users/{Redacted}/Documents/Arduino/libraries/Firmata/utility/SerialFirmata.cpp:364:56: note: suggested alternative: 'SERIAL_BUFFER_SIZE'
           if (!((bytesToRead <= 0 && bytesAvailable >= SERIAL_RX_BUFFER_SIZE/2)
                                                        ^~~~~~~~~~~~~~~~~~~~~
                                                        SERIAL_BUFFER_SIZE
```

soundanalogous, 2020-07-12:
Fixed. Thanks for the report!

soundanalogous, 2020-07-12:
Actually my fix is only temporary. If [FIRMATA_SERIAL_RX_DELAY](https://github.com/firmata/arduino/blob/master/utility/SerialFirmata.h#L56) is uncommented, the issue will still occur. The real problem is that UART_TX_FIFO_SIZE does not appear to be defined for Arduino M0 board packages, so SERIAL_RX_BUFFER_SIZE never gets defined, leading to the compile time issue pointed out here. I'll look to @jnsbyr to provide a fix. It may be an issue for other architectures supported by Firmata. I've only tested for M0 and AVR so far. There was not issue with AVR boards.

jnsbyr, 2020-07-12:
This is a platform and hardware dependent issue I am unable to resolve by
myself because my use cases are limited to AVR and ESP8266

After some diggig through the available documentation and the Arduino
source I found UART_TX_FIFO_SIZE for AVR and that it is save to asume
UART_RX_FIFO_SIZE=UART_TX_FIFO_SIZE. A similar analysis must be performed
for each platforms that where this feature is of interest.

I am currently limited to email but I think I already added a solution for
the ESP8266 using defines.

Maybe SERIAL_RX_BUFFER_SIZE needs to be defined in boards.h depending on
platform and hardware. There seems to be no universal definition available
and it is not good practice to assume any default value.