firmata/arduino #461: Send PWM value to pin ?

hoahoctro, 2020-07-30:
Im using ESP8266 as firmata master and UNO as client, they communicate via TX RX pin

I refered FirmataMashaller to control UNO pin via sendPinMode and sendDigital, but how to send Analog value ?, i tried sendAnalog(9,255) and some other values but my led seems not fade.

Hope you helps, thanks. :)

zfields, 2020-07-30:
Can you share your code?

hoahoctro, 2020-07-31:
> Can you share your code?

`#include <Boards.h>
#include <Firmata.h>
#include <FirmataConstants.h>
#include <FirmataDefines.h>
#include <FirmataMarshaller.h>
#include <FirmataParser.h>

firmata::FirmataMarshaller marshaller;

void setup() {
    Serial.begin(57600);
    marshaller.begin(Serial);
    marshaller.sendPinMode(13, OUTPUT);
    marshaller.sendPinMode(9, OUTPUT);
}

void loop () {
    marshaller.sendDigital(13, HIGH);
    marshaller.sendDigital(9, 1);
    delay(200);
    marshaller.sendDigital(13, LOW);
    marshaller.sendDigital(9, 0);
    delay(200);

    for (int fadeValue = 0 ; fadeValue <= 255; fadeValue += 5) {
    marshaller.sendAnalog(9, fadeValue);
    delay(30);
    }

    for (int fadeValue = 255 ; fadeValue >= 0; fadeValue -= 5) {
    marshaller.sendAnalog(9, fadeValue);
    delay(30);
    }
}`

zfields, 2020-07-31:
First thing I see, is that you have access to `Firmata.h`. Typically, you should be using the `Firmata.h` APIs, instead of the `FirmataMarshaller.h` APIs.

I can help you, but I may need the weekend before I have the time.

zfields, 2020-08-06:
Sorry for the delay, but I wanted to confirm my suspicions.

We have an analog example that contains both read and write. Unfortunately, they are executed in the same line, so it can be a little confusing.

Here is a link to the example:
https://github.com/firmata/arduino/blob/master/examples/SimpleAnalogFirmata/SimpleAnalogFirmata.ino#L42

The only API you should need is `Firmata.analogWrite(<pin #>, <analog value>);`. You can steer clear of invoking `FirmataMarshaller` directly, as long as you are in an Arduino environment.

zfields, 2020-08-20:
@hoahoctro Was this enough to get you going? If so, I would like to close this issue.

dtex, 2020-08-20:
@zfields Someone opened an issue on the J5 repo reporting that PWM was not working on non-PWM pins. I recreated w/o J5, using just firmata.js. <-> firmata/arduino. It's my understanding that there is some magic that allows non-PWM pins to be used as PWM pins. Not sure if this functionality comes from firmata or the Arduino libraries but am looking into it now.

Code I used to test

````js

const Board = require("firmata");

Board.requestPort((error, port) => {
  if (error) {
    console.log(error);
    return;
  }

  const board = new Board(port.path);

  board.on("ready", () => {
    const pin = 12;
    let state = 0;

    board.pinMode(pin, board.MODES.PWM);

    setInterval(() => {
      board.analogWrite(pin, state++);
      if (state === 255) state = 0;
    }, 10);
  });
});
````

dtex, 2020-08-20:
Though I see this user is on an ESP8266 so I assume 9 on there is fine for PWM so probably a different issue.

soundanalogous, 2020-08-21:
Only PWM pins can be used for PWM. There's a macro defined for each board variant, called `digitalPinHasPWM`. [Here's what that looks like](https://github.com/arduino/ArduinoCore-avr/blob/master/variants/standard/pins_arduino.h#L35) for a standard ATMega328p (e.g. Uno). Firmata [uses this macro in Boards.h](https://github.com/firmata/arduino/blob/master/Boards.h#L156) to determine which pins support PWM.

closed 2020-11-24 by soundanalogous