firmata/arduino #463: Controllino

danilogacevic, 2020-08-17:
Hi,

Is it possible to communicate with controllino using firmata.js ?

soundanalogous, 2020-08-18:
It depends on how the board presents itself. For example, if the MEGA presents itself as an Arduino Mega, or the MINI as an Arduino Uno, then it may work. Otherwise new board definitions would be added to Boards.h.

danilogacevic, 2020-08-18:
This is description for [controllno mini](https://www.controllino.biz/product/controllino-mini/). And what time would be needed for those new definitions to take place, if you know ?

Thank you

dtex, 2020-08-18:
I took a look and I think there is some functionality (and pins) that can only be accessed through the controllino.h library which augments the default Arduino libraries so depending on what you want to do, you might need a custom firmata sketch.

> Is there any other arduino based PLC that could be programmed through firmata ?

This gave me pause. Firmata is a protocol for communicating with a microcontroller and the logic of your program lives on another, host computer. Since this thread started in the Johnny-Five gitter I assumed that's what you want to do but if you are looking to program the PLC and have it be a stand-alone system, this might be the wrong path.

danilogacevic, 2020-08-18:
Yes, I hoped that I can use javascript to program controllino as stand alone system. So, as I see, the best way to accomplish this is too use Arduino IDE which is basically C++.

Thank you for explanation

closed 2020-08-18 by danilogacevic