firmata/arduino #467: If I uploaded the program once, I need to press the reset button to upload the program again

xiebin2014, 2020-09-01:
If I uploaded the program once, I need to press the reset button to upload the program again.
Is there any way to solve this problem

soundanalogous, 2020-09-01:
Which board are you using?

xiebin2014, 2020-09-01:
Thank you for your reply
Arduino Nano
ATmega328p(Old Bootloader)
StandardFirmata.ino

I think this is a common problem

soundanalogous, 2020-09-02:
Do you have this problem with other Arduino sketches or just with Firmata?

xiebin2014, 2020-09-02:
[image]
EchoString is ok

soundanalogous, 2020-09-03:
The Nano (at least since v3) has the auto reset functionality so this should not be an issue. StandardFirmata also fits on an ATmega328p so it's likely not a memory issue.

xiebin2014, 2020-12-02:
You are right
There's something wrong with my board

closed 2020-12-02 by xiebin2014