firmata/arduino #468: Add Teensy 4.1 to Boards.h

hoga85, 2020-09-11:
To run the example StandardFirmata with Teensy, it shows an error saying that the Board.h needs to be modified. If it is possible to add Teensy 4.1 to Board.h? 

I made an attempt to sort out the pins for Teensy 4.1 by following the format of that used for Teensy 3.5&3.6, the attempt is as shown below. But then I am not sure how to further make it work. Hopefully, Teensy 4.1 can be considered to added to Boards.h.

```C
// Teensy-4.1
// reference: https://github.com/PaulStoffregen/cores/blob/master/teensy4/pins_arduino.h
#elif defined(__IMXRT1062__)
#define TOTAL_ANALOG_PINS      18
#define TOTAL_PINS              55
#define VERSION_BLINK_PIN       13
#define PIN_SERIAL1_RX          0
#define PIN_SERIAL1_TX          1
#define PIN_SERIAL2_RX          7
#define PIN_SERIAL2_TX          8
#define PIN_SERIAL3_RX         15
#define PIN_SERIAL3_TX         14
#define PIN_SERIAL4_RX         16
#define PIN_SERIAL4_TX         17
#define PIN_SERIAL5_RX         21
#define PIN_SERIAL5_TX         20
#define PIN_SERIAL6_RX         25
#define PIN_SERIAL6_TX         24
#define PIN_SERIAL7_RX         28
#define PIN_SERIAL7_TX         29
#define PIN_SERIAL8_RX         34
#define PIN_SERIAL8_TX         35
#define IS_PIN_DIGITAL(p)      ((p) >= 0 && (p) <= 55) \\42 digital pins on the configuration plot, but 55 indicated in pins_arduino.h by Paul.
#define IS_PIN_ANALOG(p)       ((p) >= 14 && (p) <= 27 || (p) >= 38 && (p) <= 41 )
#define IS_PIN_PWM(p)          digitalPinHasPWM(p)
#define IS_PIN_SERVO(p)        ((p) >= 0 && (p) < MAX_SERVOS)
#define IS_PIN_I2C(p)          ((p) == 18 || (p) == 19)
#define IS_PIN_SERIAL(p)       (((p) > 13 && (p) < 18) || ((p) == 0 || (p) == 1) || ((p) == 7 || (p) == 8)|| ((p) == 24 || (p) == 25)|| ((p) == 28 || (p) == 29)|| ((p) == 34 || (p) == 35))
#define PIN_TO_DIGITAL(p)      (p)
// A0-A13 = D14-D27; A14-A17 = D38-D41;
#define PIN_TO_ANALOG(p)       ((p) <= 27 ? (p) - 14 : (p) - 24)
#define PIN_TO_PWM(p)           PIN_TO_DIGITAL(p)
#define PIN_TO_SERVO(p)        (p)
```

soundanalogous, 2020-11-23:
Support for the Teensy 4.0 was added a while back. That block could be updated to support both 4.0 and 4.1 with the following change:

```c
#elif defined(__IMXRT1062__)
#define TOTAL_ANALOG_PINS       NUM_ANALOG_INPUTS
#define TOTAL_PINS              NUM_DIGITAL_PINS
#define VERSION_BLINK_PIN       13
#define PIN_SERIAL1_RX          0
#define PIN_SERIAL1_TX          1
#define PIN_SERIAL2_RX          7
#define PIN_SERIAL2_TX          8
#define PIN_SERIAL3_RX          15
#define PIN_SERIAL3_TX          14
#define PIN_SERIAL4_RX          16
#define PIN_SERIAL4_TX          17
#define PIN_SERIAL5_RX          21
#define PIN_SERIAL5_TX          20
#define PIN_SERIAL6_RX          25
#define PIN_SERIAL6_TX          24
#define PIN_SERIAL7_RX          28
#define PIN_SERIAL7_TX          29
#define IS_PIN_DIGITAL(p)       ((p) >= 0 && (p) < NUM_DIGITAL_PINS)
#ifdef ARDUINO_TEENSY40
  #define IS_PIN_ANALOG(p)        ((p) >= 14 && (p) <= 27)
  #define PIN_TO_ANALOG(p)        ((p) - 14)
#endif
#ifdef ARDUINO_TEENSY41
  #define IS_PIN_ANALOG(p)        (((p) >= 14 && (p) <= 27) || ((p) >= 38 && (p) <= 41))
  #define PIN_TO_ANALOG(p)        (((p) <= 27) ? (p) - 14 : (p) - 24)
#endif
#define IS_PIN_PWM(p)           digitalPinHasPWM(p)
#define IS_PIN_SERVO(p)         ((p) >= 0 && (p) < MAX_SERVOS)
#define IS_PIN_I2C(p)           ((p) == PIN_WIRE_SDA || (p) == PIN_WIRE_SCL)
#define IS_PIN_SERIAL(p)        (((p) >= 0 && (p) <= 1) || ((p) >= 7 && (p) <= 8) || ((p) >= 14 && (p) <= 17) || ((p) >= 20 && (p) <= 21) || ((p) >= 24 && (p) <= 25) || ((p) >= 28 && (p) <= 29))
#define PIN_TO_DIGITAL(p)       (p)
#define PIN_TO_PWM(p)           (p)
#define PIN_TO_SERVO(p)         (p)
```

Try that and let me know if it works.

psitech, 2021-03-06:
> Try that and let me know if it works. I have not tested it as I don't have either board nor have I tried making those changes and compiling for those respective boards (that's the first place to check for issues).
>
Thanks.
Added your lines to boards.h. The StandardFirmata sketch now compiles without error on a Teensy 4.0 and a Teensy 4.1. 
I was able to turn the LED on and off remotely on both Teensy's. 
Did not check all pins and functionality though.