firmata/arduino #469: Nano 33 IoT Missing in Boards.txt

mhamilt, 2020-11-02:
The Nano 33 Iot is missing in `Boards.txt`. Given all the other SAMD21 boards are included, I'd assume there isn't anything fundamentally needing tweaked here.

As far as I can gather it just needs an addition of the following in `Boards.txt`

```cpp
// Arduino Nano 33 IoT
#elif defined(ARDUINO_SAMD_NANO_33_IOT)
#define TOTAL_ANALOG_PINS       8
#define TOTAL_PINS              21 // 13 Digital + 8 Analog
#define IS_PIN_DIGITAL(p)       (((p) >= 2 && (p) <= 13))
#define IS_PIN_ANALOG(p)        ((p) >= 14 && (p) < 22)
#define IS_PIN_PWM(p)           digitalPinHasPWM(p)
#define IS_PIN_SERVO(p)         (IS_PIN_DIGITAL(p) && (p) < MAX_SERVOS) // deprecated since v2.4
#define IS_PIN_I2C(p)           ((p) == SDA || (p) == SCL)
#define IS_PIN_SPI(p)           ((p) == SS || (p) == MOSI || (p) == MISO || (p) == SCK)
#define IS_PIN_SERIAL(p)        ((p) == PIN_SERIAL1_RX || (p) == PIN_SERIAL1_TX) //defined in variant.h  RX = 13, TX = 14
#define PIN_TO_DIGITAL(p)       (p)
#define PIN_TO_ANALOG(p)        ((p) - 15)
#define PIN_TO_PWM(p)           PIN_TO_DIGITAL(p)
#define PIN_TO_SERVO(p)         (p) // deprecated since v2.4
```

tommmmudd, 2020-11-12:
Note that this is the Boards.h rather than Boards.txt that needs updating.
To edit this file with the above text, do the following:

- Open the Arduino software
- Go to File --> Examples --> Firmata --> StandardFirmata
- Press Cmd-k (ctrl-k on Windows), or go to Sketch --> Show Sketch Folder

This should open finder/explorer to a location like Firmata/examples/StandardFirmata, you then need to do the following:
- Go up two folder levels to the Firmata folder (cmd-up on a Mac)
- You should then see the Boards.h file. Open this (e.g. in TextEdit or Notepad)
- Copy Matthew's code above to the clipboard
- Scroll down in the Boards.h file to the bottom, then scroll back up looking for the line "// anything else"
- Paste Matthew's code on the line aboce the // anything else, so the file should look something like this:

`
...
#define PIN_TO_PWM(p)           (p)
#define PIN_TO_SERVO(p)         (p)

// Arduino Nano 33 IoT
#elif defined(ARDUINO_SAMD_NANO_33_IOT)
...
 (etc, with the rest of the pasted code). 
...
#define PIN_TO_SERVO(p)         (p) // deprecated since v2.4

// anything else
#else
#error "Please edit Boards.h with a hardware abstraction for this board"
...
`

soundanalogous, 2020-11-22:
Have you tested this with a Nano 33?

soundanalogous, 2020-11-23:
Out of curiosity, what are you trying to accomplish using Firmata when the Nano 33 IoT board is already compatible with several Cloud services like Arduino IoT Cloud, Firebase, Azure, etc? As more boards are powerful enough to interact with Cloud Services, yet priced the same or less than boards like an Arduino Zero, I'd think the primary use case remaining for Firmata is simply using it over a UART.

mhamilt, 2020-11-23:
@soundanalogous 

>Have you tested this with a Nano 33?

Yes, and everything seems to work, however I have not tested it extensively.

> I'd think the primary use case remaining for Firmata is simply using it over a UART for quick iterative prototyping.

Correct, I am also using it for teaching purposes where students' first interaction with the Nano 33 will primarily be Serial comms based.

Since the MKR boards are already included in `Boards.h` 

https://github.com/firmata/arduino/blob/d4cde66bdc0f7813a03bc296ec80a899c32d412a/Boards.h#L264-L265

and since there does not seem to be any adverse effects on the Firmata itself, I could not see the harm in including the Nano 33 IoT.

soundanalogous, 2020-11-29:
I've added this board per https://github.com/firmata/arduino/pull/474, but do not have the physical board to verify with. I've only ensured it will compile StandardFirmata. Please let me know if you encounter any issues.

closed 2020-11-29 by soundanalogous

bilal697, 2021-09-12:
#error "Please edit Boards.h with a hardware abstraction for this board"

Hello Anyone assist me to solve this error for Arduino Nano 33 BLE Sense.

cstawarz, 2021-10-30:
@agentff6600: I just submitted a PR (#485) that adds support for ArduinoBLE-compatible boards.  My test device was the Nano 33 IoT.   Be sure to install ArduinoBLE via the Library Manager before compiling.