firmata/arduino #470: multiple issues when compiling StandardFirmataWiFi for MKR 1010

Jorgeruiz97, 2020-11-21:
# Version
*Firmata*: latest from master
*Arduino IDE*: Arduino: 1.8.14 Hourly Build 2020/09/23 10:35 (Mac OS X)
*Board*: "Arduino MKR WiFi 1010"
*OS version*: macOS Big Sur 11.0.1

# What am I trying to achieve?
I'm trying to compile the StandardFirmataWiFi.ino

# Issue
When trying to compile the StandardFirmataWiFi.ino I'm getting the following errors. The StandardFirmata.ino example works and compiles fine.

```bash
Compiling sketch...
/Users/jorgesmacbookpro/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD -DF_CPU=48000000L -DARDUINO=10814 -DARDUINO_SAMD_MKRWIFI1010 -DARDUINO_ARCH_SAMD -DUSE_ARDUINO_MKR_PIN_LAYOUT -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x8054 -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino MKR WiFi 1010\"" -DUSE_BQ24195L_PMIC -I/Users/jorgesmacbookpro/Library/Arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/ -I/Users/jorgesmacbookpro/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/ -I/Users/jorgesmacbookpro/Library/Arduino15/packages/arduino/hardware/samd/1.8.9/cores/arduino -I/Users/jorgesmacbookpro/Library/Arduino15/packages/arduino/hardware/samd/1.8.9/variants/mkrwifi1010 -I/Applications/Arduino.app/Contents/Java/libraries/Servo/src -I/Users/jorgesmacbookpro/Library/Arduino15/packages/arduino/hardware/samd/1.8.9/libraries/Wire -I/Users/jorgesmacbookpro/Documents/Arduino/libraries/Firmata -I/Users/jorgesmacbookpro/Documents/Arduino/libraries/WiFiNINA/src -I/Users/jorgesmacbookpro/Library/Arduino15/packages/arduino/hardware/samd/1.8.9/libraries/SPI /var/folders/v6/1h477kz97qx76rglxdhffm4c0000gn/T/arduino_build_330385/sketch/StandardFirmataWiFi.ino.cpp -o /var/folders/v6/1h477kz97qx76rglxdhffm4c0000gn/T/arduino_build_330385/sketch/StandardFirmataWiFi.ino.cpp.o
In file included from /Users/jorgesmacbookpro/Documents/Arduino/libraries/Firmata/Firmata.h:18:0,
                 from /Users/jorgesmacbookpro/Desktop/iot/StandardFirmataWiFi/StandardFirmataWiFi.ino:79:
/Users/jorgesmacbookpro/Documents/Arduino/libraries/Firmata/FirmataDefines.h:67:33: error: 'firmata' redeclared as different kind of symbol
 #define SET_PIN_MODE            firmata::SET_PIN_MODE // set a pin to INPUT/OUTPUT/PWM/etc
                                 ^
/Users/jorgesmacbookpro/Documents/Arduino/libraries/WiFiNINA/src/utility/wifi_spi.h:102:5: note: in expansion of macro 'SET_PIN_MODE'
     SET_PIN_MODE  = 0x50,
     ^~~~~~~~~~~~
In file included from /Users/jorgesmacbookpro/Documents/Arduino/libraries/Firmata/FirmataDefines.h:17:0,
                 from /Users/jorgesmacbookpro/Documents/Arduino/libraries/Firmata/Firmata.h:18,
                 from /Users/jorgesmacbookpro/Desktop/iot/StandardFirmataWiFi/StandardFirmataWiFi.ino:79:
/Users/jorgesmacbookpro/Documents/Arduino/libraries/Firmata/FirmataConstants.h:17:11: note: previous declaration 'namespace firmata { }'
 namespace firmata {
           ^~~~~~~
In file included from /Users/jorgesmacbookpro/Documents/Arduino/libraries/Firmata/Firmata.h:18:0,
                 from /Users/jorgesmacbookpro/Desktop/iot/StandardFirmataWiFi/StandardFirmataWiFi.ino:79:
/Users/jorgesmacbookpro/Documents/Arduino/libraries/Firmata/FirmataDefines.h:67:40: error: expected '}' before '::' token
 #define SET_PIN_MODE            firmata::SET_PIN_MODE // set a pin to INPUT/OUTPUT/PWM/etc
                                        ^
/Users/jorgesmacbookpro/Documents/Arduino/libraries/WiFiNINA/src/utility/wifi_spi.h:102:5: note: in expansion of macro 'SET_PIN_MODE'
     SET_PIN_MODE  = 0x50,
     ^~~~~~~~~~~~
/Users/jorgesmacbookpro/Documents/Arduino/libraries/Firmata/FirmataDefines.h:67:42: error: explicit qualification in declaration of 'SET_PIN_MODE'
 #define SET_PIN_MODE            firmata::SET_PIN_MODE // set a pin to INPUT/OUTPUT/PWM/etc
                                          ^
/Users/jorgesmacbookpro/Documents/Arduino/libraries/WiFiNINA/src/utility/wifi_spi.h:102:5: note: in expansion of macro 'SET_PIN_MODE'
     SET_PIN_MODE  = 0x50,
     ^~~~~~~~~~~~
In file included from /Users/jorgesmacbookpro/Documents/Arduino/libraries/WiFiNINA/src/utility/wifi_drv.h:25:0,
                 from /Users/jorgesmacbookpro/Documents/Arduino/libraries/WiFiNINA/src/WiFiStorage.h:23,
                 from /Users/jorgesmacbookpro/Documents/Arduino/libraries/WiFiNINA/src/WiFi.h:37,
                 from /Users/jorgesmacbookpro/Documents/Arduino/libraries/WiFiNINA/src/WiFiNINA.h:23,
                 from /var/folders/v6/1h477kz97qx76rglxdhffm4c0000gn/T/arduino_build_330385/sketch/wifiConfig.h:130,
                 from /Users/jorgesmacbookpro/Desktop/iot/StandardFirmataWiFi/StandardFirmataWiFi.ino:98:
/Users/jorgesmacbookpro/Documents/Arduino/libraries/WiFiNINA/src/utility/wifi_spi.h:102:21: error: invalid conversion from 'int' to '<unnamed enum>' [-fpermissive]
     SET_PIN_MODE  = 0x50,
                     ^~~~
/Users/jorgesmacbookpro/Documents/Arduino/libraries/WiFiNINA/src/utility/wifi_spi.h:103:25: error: invalid conversion from 'int' to '<unnamed enum>' [-fpermissive]
     SET_DIGITAL_WRITE = 0x51,
                         ^~~~
/Users/jorgesmacbookpro/Documents/Arduino/libraries/WiFiNINA/src/utility/wifi_spi.h:104:24: error: invalid conversion from 'int' to '<unnamed enum>' [-fpermissive]
     SET_ANALOG_WRITE = 0x52,
                        ^~~~
/Users/jorgesmacbookpro/Documents/Arduino/libraries/WiFiNINA/src/utility/wifi_spi.h:107:20: error: invalid conversion from 'int' to '<unnamed enum>' [-fpermissive]
     WRITE_FILE   = 0x60,
                    ^~~~
/Users/jorgesmacbookpro/Documents/Arduino/libraries/WiFiNINA/src/utility/wifi_spi.h:108:19: error: invalid conversion from 'int' to '<unnamed enum>' [-fpermissive]
     READ_FILE   = 0x61,
                   ^~~~
/Users/jorgesmacbookpro/Documents/Arduino/libraries/WiFiNINA/src/utility/wifi_spi.h:109:21: error: invalid conversion from 'int' to '<unnamed enum>' [-fpermissive]
     DELETE_FILE   = 0x62,
                     ^~~~
/Users/jorgesmacbookpro/Documents/Arduino/libraries/WiFiNINA/src/utility/wifi_spi.h:110:21: error: invalid conversion from 'int' to '<unnamed enum>' [-fpermissive]
     EXISTS_FILE   = 0x63,
                     ^~~~
/Users/jorgesmacbookpro/Documents/Arduino/libraries/WiFiNINA/src/utility/wifi_spi.h:111:22: error: invalid conversion from 'int' to '<unnamed enum>' [-fpermissive]
     DOWNLOAD_FILE  = 0x64,
                      ^~~~
/Users/jorgesmacbookpro/Documents/Arduino/libraries/WiFiNINA/src/utility/wifi_spi.h:112:25: error: invalid conversion from 'int' to '<unnamed enum>' [-fpermissive]
     APPLY_OTA_COMMAND = 0x65,
                         ^~~~
/Users/jorgesmacbookpro/Documents/Arduino/libraries/WiFiNINA/src/utility/wifi_spi.h:113:18: error: invalid conversion from 'int' to '<unnamed enum>' [-fpermissive]
  RENAME_FILE   = 0x66,
                  ^~~~
/Users/jorgesmacbookpro/Documents/Arduino/libraries/WiFiNINA/src/utility/wifi_spi.h:114:18: error: invalid conversion from 'int' to '<unnamed enum>' [-fpermissive]
  DOWNLOAD_OTA  = 0x67,
                  ^~~~
/Users/jorgesmacbookpro/Documents/Arduino/libraries/WiFiNINA/src/utility/wifi_spi.h:115:1: error: expected unqualified-id before '}' token
 };
 ^
/Users/jorgesmacbookpro/Documents/Arduino/libraries/WiFiNINA/src/utility/wifi_spi.h:115:1: error: expected declaration before '}' token
Using library Servo at version 1.1.6 in folder: /Applications/Arduino.app/Contents/Java/libraries/Servo 
Using library Wire at version 1.0 in folder: /Users/jorgesmacbookpro/Library/Arduino15/packages/arduino/hardware/samd/1.8.9/libraries/Wire 
Using library Firmata at version 2.5.8 in folder: /Users/jorgesmacbookpro/Documents/Arduino/libraries/Firmata 
Using library WiFiNINA at version 1.8.0 in folder: /Users/jorgesmacbookpro/Documents/Arduino/libraries/WiFiNINA 
Using library SPI at version 1.0 in folder: /Users/jorgesmacbookpro/Library/Arduino15/packages/arduino/hardware/samd/1.8.9/libraries/SPI 
exit status 1
Error compiling for board Arduino MKR WiFi 1010.
```

soundanalogous, 2020-11-30:
It appears the issue is both Firmata and [WiFiNINA](https://github.com/arduino-libraries/WiFiNINA/blob/master/src/utility/wifi_spi.h#L102) are defining `SET_PIN_MODE`, but as different types.

zfields, 2020-11-30:
I'm looking into it. Let me see if I can come up with something.

soundanalogous, 2020-11-30:
FWIW, the issue here appears to be in WiFiNINA, not Firmata.  That enum should be restricted in scope.

zfields, 2020-11-30:
In all fairness, it takes two to clash, and we are also defining it in the global scope. :smile: If I recall correctly, we were only left it in the global scope for the sake of backward compatibility, and there are no longer any dependencies in Firmata. Furthermore, it is not necessary any new programs written using Firmata.

To move forward, I think you can safely comment out line 67 in the file `/Users/jorgesmacbookpro/Documents/Arduino/libraries/Firmata/FirmataDefines.h`.

soundanalogous, 2020-11-30:
That compiles but I think `SET_PIN_MODE` holds the value declared in WiFiNINA rather than in FirmataConstants, since the firmata namespace is not declared in any of the example files. Adding the firmata namespace brings up some other issues, but they're resolvable.

zfields, 2020-11-30:
Yes, that would be my expectation. That is probably the bug here, is that the samples are not `using namespace firmata;`.

I can take some time over this coming week, and clean up the samples to use the using command or the namespaced constants where applicable.

Jorgeruiz97, 2020-11-30:
Thank you guys! I will try it out during the week and will let you know.