firmata/arduino #471: Identify each Firmata-flashed board with a unique name

PizzaProgram, 2020-11-22:
After 2 years I still couldn't find any standard solution for naming the boards before flashing and searching for unique names to identify which board is at which usb/ip.  
  
At sample .INO files there should be something like this at the beginning:  
`#define FIRMATA_UNIQE_BOARD_NAME "nano_b126_living_room_temp"; // <<-- RENAME this before flashing !! `
...
And the identification could be something like:
` string USBport = firmata.SearchSerialPortbyName("Mega36_basement");`
So it would be nice to implement a simple search algorithm to help implement this basic feature to each programming language adapting Firmata.

Also a simple readout, like:  
`if (firmata.ReadBoardName() != "b123_garage_door_relay") return;` 
would help a lot to prevent any accidents.

Why is this great protocol supports only 1 board by default? 
 _Mixing up a board with others, if the USB port or  IP number changes makes this protocol dangerous to use at real life or industrial enviroments._

soundanalogous, 2020-11-22:
The protocol supports several MCU architectures by default. The reason each board is not uniquely identified is this does not scale well and would be a huge maintenance issue. See [boards.h](https://github.com/firmata/arduino/blob/master/Boards.h) for all of the currently supported architectures. Each board reports is features using the [capability query](https://github.com/firmata/protocol/blob/master/protocol.md#capability-query).

Also, Firmata is not engineered for use in "industrial environments", as an open source project, it's "use at your own risk".

closed 2020-11-22 by soundanalogous

soundanalogous, 2020-11-22:
There is also a simple solution to what I believe you are trying to accomplish, that is to name the .ino file something unique for each board. So rather than StandardFirmata.ino, "boardA.ino", "boardB.ino", etc. There is no way to name the serial port from the code running on an Arduino that I'm aware of so, you'd have to attempt to make a connection on each serial port, and if a board running Firmata is found, then you can get the exact board based on what you named the .ino file.

PizzaProgram, 2020-11-23:
> The reason each board is not uniquely identified is this does not scale well and would be a huge maintenance issue.
  
I think you have misunderstood me a bit. My suggestion has nothing to with the fact, how many **kinds** of board Firmata supports. (Which is extremly cool by the way :1st_place_medal: )  
I'm talking about a different / independent naming.  

> So rather than StandardFirmata.ino, name the sketches: "boardA.ino", "boardB.ino"

This solution sounds interesting! :)  
And what **command** is it, that gives me back the name of the "boardB.ino" from the host side?
Basically how do I "ask" the board connected currently to "/dev/ttyUSB3" >> Who are you? What "boardname" do you have?

soundanalogous, 2020-11-23:
Use the query firmware command [as described here](https://github.com/firmata/protocol/blob/master/protocol.md#query-firmware-name-and-version).

dtex, 2020-11-23:
@PizzaProgram Jeff's suggestion works well. I had a project that used multiple boards of the same type (Uno's). Renaming the sketch allowed me to sniff out which board I had connected to at each address.

PizzaProgram, 2020-12-01:
@soundanalogous You are right, it's already done! I was "knocking on opened doors".
Maybe 1-1 sentence comment on the front page + in the SandardFirmata.ino sample file would save others many time too.
I'm looking at :
file:///C:/Program%20Files%20(x86)/Arduino/libraries/Firmata/docs/html/index.html

Now I have to find a way to turn this into a REAL example, how to do it from Node-Red.
 (Learning and trying since a week without success.)

PizzaProgram, 2020-12-01:
> Jeff's suggestion works well. I had a project that used multiple boards of the same type (Uno's). Renaming the sketch allowed me to sniff out which board I had connected to at each address.

@dtex Many thanks for the feedback!
Can you please share your code, how you did it Please?

dtex, 2020-12-02:
I can't help with node-red, but with firmata.js you just check the ```board.name``` property. It will be whatever you named the sketch that was uploaded to that board. Does node-red expose the board object?