firmata/arduino #475: add seeedunio xiao to boards.h please

robobibble, 2020-12-04:
hi could you add the seeeduino xiao to boards.h please

thanks
lee

Vicckkky, 2021-02-08:
Hey,

You can add the following to boards.h to support the XIAO.

Cheers,

// Seeduino XIAO

#elif defined(SEEED_XIAO_M0)
#define TOTAL_ANALOG_PINS       11
#define TOTAL_PINS              17 // 11 digital / analog + 1 DAC output + 2 i2c + 3 spi
#define VERSION_BLINK_PIN       LED_BUILTIN
#define PIN_SERIAL1_RX          7
#define PIN_SERIAL1_TX          6
#define IS_PIN_DIGITAL(p)       ((p) >= 0 && (p) <= 10)
#define IS_PIN_ANALOG(p)        ((p) >= 0 && (p) <= 10)
#define IS_PIN_PWM(p)           ((p) >= 1 && (p) <= 10)
#define IS_PIN_SERVO(p)         (IS_PIN_DIGITAL(p) && (p) < MAX_SERVOS) // deprecated since v2.4
#define IS_PIN_I2C(p)           ((p) == 4 || (p) == 5) // SDA = 4, SCL = 5
#define IS_PIN_SPI(p)           ((p) == 4 || (p) == 10 || (p) == 9 || (p) == 8) // SS = A4
#define IS_PIN_SERIAL(p)        ((p) == 6 || (p) == 7)
#define PIN_TO_DIGITAL(p)       (p)
#define PIN_TO_ANALOG(p)        (p)
#define PIN_TO_PWM(p)           PIN_TO_DIGITAL(p)
#define PIN_TO_SERVO(p)         (p) // deprecated since v2.4

gerhard1951, 2021-08-11:
Hi,
the above mentioned code works perfectly for the Seeeduino XIAO, which was added in the boards.h file.
However, the code was not detected by the "elif" line, showing the name of the Seeduino. I tried it with (SEEED_XIAO_M0) and (Seeeduino_XIAO) and with or without the _M0.
Finally I prepared a special boards.h file, replacing "elif" before "define" with "else". This worked, because only the parameters of the Seeeduino Board were used. But with this trick I need for every board type a manipulated boards.h file
What am I doing wrong, that the right board ist not detected automatically? Any idea?
Thanks
Gerhard

pgrawehr, 2021-08-11:
You might find the correct name to use in the compiler output. Look in the compiler output for a parameter that starts with -D. You can also post the compiler output here if you don't find it. Make sure the output is set to verbose (in the IDE preferences).

gerhard1951, 2021-08-12:
Thanks! I found the correct name in the compiler output: "ARDUINO-SEEED-XIAO-M0" works!

pgrawehr, 2021-08-12:
Would be great if you could create a PR for this, but preferably on ConfigurableFirmata, as this repo is not being actively maintained at this time.

zfields, 2021-08-13:
To clarify, we would appreciate your contribution to this library. It is being maintained, but not advanced.

Any contributions here will help all those using Firmata via the Arduino IDE.

gerhard1951, 2021-08-14:
I am willing to contribute, but never before I did a PR. I have to learn first how to make a PR.

My Seeduino board worked with the following definition:
// Seeeduino XIAO
#elif defined(ARDUINO_SEEED_XIAO_M0)

My NANO EVERY board as follows:
//ARDUINO_NANO_EVERY Board   
#elif defined(ARDUINO_AVR_NANO_EVERY )

al-davis, 2024-02-13:
I'm not having luck with this.  I copied & pasted, to the end of the existing boards list,  Vickkkkys block along with gerhard1951 corrected name to my board.h that I brought local to my project for testing.  I continue to get several warnings and error : Please edit Boards.h with a hardware abstraction for this board.

pgrawehr, 2024-02-13:
@al-davis Please post the build log (make sure to enable verbose output first). That will help in understanding the problem.

al-davis, 2024-02-13:
Arduino: 1.8.19 (Windows Store 1.8.57.0) (Windows 10), Board: "Seeeduino XIAO, Arduino, Off"

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\User\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\User\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\User\Documents\Arduino\libraries -fqbn=Seeeduino:samd:seeed_XIAO_m0:usbstack=arduino,debug=off -ide-version=10819 -build-path C:\Users\User\AppData\Local\Temp\arduino_build_859572 -warnings=default -build-cache C:\Users\User\AppData\Local\Temp\arduino_cache_231637 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=C:\Users\User\Documents\ArduinoData\packages\Seeeduino\tools\bossac\1.8.0-48-gb176eee -prefs=runtime.tools.bossac-1.7.0-arduino3.path=C:\Users\User\Documents\ArduinoData\packages\Seeeduino\tools\bossac\1.7.0-arduino3 -prefs=runtime.tools.arduinoOTA.path=C:\Users\User\Documents\ArduinoData\packages\Seeeduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Users\User\Documents\ArduinoData\packages\Seeeduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\User\Documents\ArduinoData\packages\Seeeduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\User\Documents\ArduinoData\packages\Seeeduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.bossac-1.8.0-48-gb176eee.path=C:\Users\User\Documents\ArduinoData\packages\Seeeduino\tools\bossac\1.8.0-48-gb176eee -prefs=runtime.tools.CMSIS.path=C:\Users\User\Documents\ArduinoData\packages\Seeeduino\tools\CMSIS\5.4.0 -prefs=runtime.tools.CMSIS-5.4.0.path=C:\Users\User\Documents\ArduinoData\packages\Seeeduino\tools\CMSIS\5.4.0 -prefs=runtime.tools.CMSIS-Atmel.path=C:\Users\User\Documents\ArduinoData\packages\Seeeduino\tools\CMSIS-Atmel\1.2.1 -prefs=runtime.tools.CMSIS-Atmel-1.2.1.path=C:\Users\User\Documents\ArduinoData\packages\Seeeduino\tools\CMSIS-Atmel\1.2.1 -prefs=runtime.tools.openocd.path=C:\Users\User\Documents\ArduinoData\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=C:\Users\User\Documents\ArduinoData\packages\arduino\tools\openocd\0.10.0-arduino7 -verbose C:\Users\User\Documents\Arduino\WaveShareDemoMin\WaveShareDemoMin.ino

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\User\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\User\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\User\Documents\Arduino\libraries -fqbn=Seeeduino:samd:seeed_XIAO_m0:usbstack=arduino,debug=off -ide-version=10819 -build-path C:\Users\User\AppData\Local\Temp\arduino_build_859572 -warnings=default -build-cache C:\Users\User\AppData\Local\Temp\arduino_cache_231637 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=C:\Users\User\Documents\ArduinoData\packages\Seeeduino\tools\bossac\1.8.0-48-gb176eee -prefs=runtime.tools.bossac-1.7.0-arduino3.path=C:\Users\User\Documents\ArduinoData\packages\Seeeduino\tools\bossac\1.7.0-arduino3 -prefs=runtime.tools.arduinoOTA.path=C:\Users\User\Documents\ArduinoData\packages\Seeeduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Users\User\Documents\ArduinoData\packages\Seeeduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\User\Documents\ArduinoData\packages\Seeeduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\User\Documents\ArduinoData\packages\Seeeduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.bossac-1.8.0-48-gb176eee.path=C:\Users\User\Documents\ArduinoData\packages\Seeeduino\tools\bossac\1.8.0-48-gb176eee -prefs=runtime.tools.CMSIS.path=C:\Users\User\Documents\ArduinoData\packages\Seeeduino\tools\CMSIS\5.4.0 -prefs=runtime.tools.CMSIS-5.4.0.path=C:\Users\User\Documents\ArduinoData\packages\Seeeduino\tools\CMSIS\5.4.0 -prefs=runtime.tools.CMSIS-Atmel.path=C:\Users\User\Documents\ArduinoData\packages\Seeeduino\tools\CMSIS-Atmel\1.2.1 -prefs=runtime.tools.CMSIS-Atmel-1.2.1.path=C:\Users\User\Documents\ArduinoData\packages\Seeeduino\tools\CMSIS-Atmel\1.2.1 -prefs=runtime.tools.openocd.path=C:\Users\User\Documents\ArduinoData\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=C:\Users\User\Documents\ArduinoData\packages\arduino\tools\openocd\0.10.0-arduino7 -verbose C:\Users\User\Documents\Arduino\WaveShareDemoMin\WaveShareDemoMin.ino

[... cut ...]
Using library Adafruit_ZeroDMA at version 1.0.4 in folder: C:\Users\User\Documents\ArduinoData\packages\Seeeduino\hardware\samd\1.8.1\libraries\Adafruit_ZeroDMA 

Using library Firmata at version 2.5.8 in folder: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\libraries\Firmata 

Using library WaveShape_4in2 in folder: C:\Users\User\Documents\Arduino\libraries\WaveShape_4in2 (legacy)

exit status 1

#error "Please edit Boards.h with a hardware abstraction for this board"

pgrawehr, 2024-02-13:
Hmm.... That should work. Maybe you edited the wrong file. The file you are suposed to edit is `C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\libraries\Firmata\Boards.h`

al-davis, 2024-02-13:
Well, that is the file I started with.  I brought a copy into the same space as the .ino file and edited it.  Also changed the #include in the .ino file from #include <Boards.h> to #include "Boards.h" .  Now I'm wondering is some of the firma files are getting the original boards.h.  I'll go check into that.

al-davis, 2024-02-13:
Yes that is an issue.  Now I'm trying to sort thru  permissions hell.  My projects/IDE  aren't in Administrator space but all these files are.  Thanks for your help, I'll let you know.

al-davis, 2024-02-13:
OK, that got it to compile.  I appreciate you jumping in to help.

Al