firmata/arduino #489: Pyfirmata and arduino leonardo mouse and keyboard abilities

vexzyfr, 2022-03-02:
I've been dying to figure this out. I'm very inexperienced with microcontrollers so please pardon my cluelessness.
In the arduino IDE the arduino leonardo has the capability to take control of your mouse and keyboard and emulate clicks, movements, and button presses for both. I'm not too sure as for how to use pyfirmata yet, but I haven't found any info on using the equivalent of mouse.move() and other related functions with pyfirmata, is this possible, and if so, how? I do have a usb host shield at hand if needed.
Thanks!

pgrawehr, 2022-03-02:
@vexzyfr Firmata is intended to control hardware that's attached to the arduino from your PC. What you're trying to do is control the PC from the Arduino. Of course it's possible to create an extension to firmata (or rather, ConfigurableFirmata) to use the keyboard/mouse interfaces to do stuff, but this seems a bit pointless. Why would you want to use the PC to tell the Arduino to hit keys so that the PC does something? 

What is it you're trying to do at the end of the day?

vexzyfr, 2022-03-02:
I won't bother you with the details as the project is relatively pointless as you said, I'm just really determined to figure this out. Thanks for the reply, I found a solution using pyserial.

closed 2022-03-02 by vexzyfr