firmata/arduino #493: Error compiling Firmata for board Arduino Nano Every

ThorPrez, 2022-05-14:
Hi @all

Now it´s 2022 and there´s still a problem with Firmata. I got my Nano Every today and have changed the boards.h.
I´m trying it with:
elif defined(AVR_NANO_EVERY) || elif defined(ARDUINO_NANO_EVERY)
and getting this fault:
In file included from C:\Program Files (x86)\Arduino\libraries\Firmata/Firmata.h:17:0,
from C:\Users\ARBEIT~1\AppData\Local\Temp\arduino_modified_sketch_434776\StandardFirmata.ino:28:
C:\Program Files (x86)\Arduino\libraries\Firmata/Boards.h:870:2: error: #error "Please edit Boards.h with a hardware abstraction for this board"
#error "Please edit Boards.h with a hardware abstraction for this board"

Is there actually a working solution ?

Thanks,
Thorsten

pgrawehr, 2022-05-15:
Hi @ThorPrez 

Please do the following: In the Arduino IDE, go to File->Preferences and enable verbose mode for compiling. Then run a build and copy the whole build output to a file. Attach this file here. The build output will contain the exact name of the preprocessor definition that is to be used.

ThorPrez, 2022-05-15:
Thanks for your feedback.
Where exactly can I see the right name for the board ?
Here are some examples of the output;

> Arduino: 1.8.19 (Windows 10), Board: "Arduino Nano Every, None (ATMEGA4809)"

> Using board 'nona4809' from platform in folder: C:\Users\Arbeitszimmer\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.7
> Using core 'arduino' from platform in folder: C:\Users\Arbeitszimmer\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.7

> Detecting libraries used...
> .... mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_NANO_EVERY -DARDUINO_ARCH_MEGAAVR ....

pgrawehr, 2022-05-15:
It's the parameters prefixed with "-D" (excluding the D itself). So try 
```
elif defined(AVR_NANO_EVERY) || defined(ARDUINO_NANO_EVERY) || defined(ARDUINO_AVR_NANO_EVERY)
```

pgrawehr, 2022-05-15:
Make sure you don't miss the `#` before `elif`

ThorPrez, 2022-05-15:
With # before Elif I'm having more faults

pgrawehr, 2022-05-15:
Then you probably have a syntax error. Tre the line exactly as I specified above.

ThorPrez, 2022-05-15:
> #elif defined(AVR_NANO_EVERY) || defined(ARDUINO_NANO_EVERY) || defined(ARDUINO_AVR_NANO_EVERY)

it works, thanks

pgrawehr, 2022-05-29:
I have updated the source and after fixing the typo this works, so closing the issue.

closed 2022-05-29 by pgrawehr