firmata/arduino #494: Error when compiling StandardFirmata 2.5.8 for Ardbox Analog HF + Wifi from Industrialshields

SKH2800, 2022-06-29:
Good morning,

I wanted to have a USB communication between the Arduino controller and a Raspberry or a PC under Windows. The PLC contains an Arduino Leonardo and when I want to compile the StandardFirmata version 2.5.8 sketch with the Arduino IDE the following error appears: 

[image]

 I finally found a temporary solution that is to load a previous version of Firmata (version 2.5.0). If anyone have a better solution, I will appreciate to know it.

Thanks in advance.

pgrawehr, 2022-07-14:
I suggest to build directly from the latest master version (clone the repository to the ~/Arduino/libraries/ folder) and try again. For new projects, I recommend to use ConfigurableFirmata, as it is more flexible and includes I2C and SPI support. See instructions in the repository there for cloning.

SKH2800, 2022-11-03:
Thank you for your answer and sorry for the late answer. I had this reply from the manufacturer :

if you are using the Ardbox Analog WiFi & BLE board to compile and upload, Serial1 port is used internally for the WiFi & BLE module (internal ESP32) so that, Serial1 port is named as wifi (wifi.begin(), wifi.println(), ...

I didn't tried since then but that seems to be the answer.

closed 2022-11-03 by SKH2800