firmata/arduino #497: Last released version is from 2018 despite a lot of new work being merged

mphasize, 2022-12-13:
I was about to create an issue for Arduino MKR WiFi support, but then found out that it was already added via a PR in 2019. There seems to also be a lot of other new work done, however the last official release is version 2.5.8 from 2018.

Since the recommended way to install the Firmata Library in the Arduino IDE is to go through the library manager and since the Library Manager can only show released versions, I'm hoping that a new release would be possible.

Is there anything that I'm missing or is there a reason that all the new work is not being released?

pgrawehr, 2022-12-13:
Hi @mphasize 

You are right, there have not been any official releases in a long time. Unfortunately, the owner of the project, @soundanalogous has not had much time to put effort into it, which also explains that most PR's are untouched for a long time. 

I have been somewhat taking over the support of this project, but I'm concentrating on ConfigurableFirmata, which is more flexible and supported also on newer boards. However, I don't know how to officially publish a new release (nor do I have the permission to do so, I suppose). 

I recommend to install the library manually using git, it's almost as easy as trough the library manager: Locate the Arduino library folder (on Windows, it's `C:\Users\<your username>\Documents\Arduino\libraries`, on linux it's `~/Arduino/libraries`). If already present, delete the library (subdirectories `arduino` or `ConfigurableFirmata` respectively) and then clone the library directly from github into that folder. I shall update the documentation to make these instructions easier to find.

soundanalogous, 2022-12-13:
@pgrawehr there are 2 types of releases:

1. Arduino releases, which will appear in the Arduino library manager. Updating this is as simple as updating the version number in the `library.properties` file just as you have done [here for ConfigurableFirmata](https://github.com/firmata/ConfigurableFirmata/blob/master/library.properties).
2. Github releases. First you have to create a git tag, then once you have pushed the tag to the repo, use the [Github release UI](https://github.com/firmata/ConfigurableFirmata/releases/new) to publish a new pre-release or latest release based on a git tag and branch. Let me know if you don't have access to that UI. I used to also attached a couple of zips with each release but that is not really necessary. I also included release notes, which can be helpful.

mphasize, 2022-12-13:
Thanks for the quick and helpful replies. Please let me know if I can help with preparing a new release.

pgrawehr, 2022-12-13:
@soundanalogous I was talking about releases for the Arduino library manager (I know how to do github releases, I have done that for other projects already). I'll need to check whether I have the permission to do so. 

It seems I'm lacking something to do an Arduino release though. As you already noticed, the master version of ConfigurableFirmata is at v2.11, but when I look at the library manager in the Arduino IDE, I still get only 2.10.1 offered. Do I still need to create the tag or something?

soundanalogous, 2022-12-13:
@pgrawehr it looks like a tag is necessary in addition to updating the version in the library.properties file. From the [Arduino library documentation](https://github.com/arduino/library-registry#readme):

> Create a [release](https://docs.github.com/repositories/releasing-projects-on-github/managing-releases-in-a-repository) or [tag](https://git-scm.com/docs/git-tag). The Library Manager index always uses tagged versions of the libraries, so even if the development version of the library is compliant, it can't be accepted until the latest release or tag is compliant. Alternatively, you can redo the existing release/tag if you prefer.

Looks like they have added a lint check as of 2021 so it's possible ArduinoBot may reply if there are any issues. That link above has all the detail. firmata/ConfigurableFirmata and firmata/Arduino are both already registered so there is no need for that initial step.

soundanalogous, 2022-12-14:
I've pushed a 2.5.9 tag to firmata/arduino so it should be available in the Arduino IDE once the bot takes it's next pass... as long as there are no lint issues.

mphasize, 2022-12-22:
Thanks everyone for looking into this. Unfortunately it looks like there's still a small piece missing. Although I can see the 2.5.9 release here on Github, the Library Manager in the Arduino IDE hasn't picked up on the new version yet. Any ideas what needs to be done there?

pgrawehr, 2022-12-28:
I have now run the arduino-linter tool which they recommend for testing compliance. I re-released the update after a fix, so hopefully it works now.

pgrawehr, 2023-03-29:
Firmata Version 2.5.9 has been released and is available for download using the library manager.