firmata/arduino #504: Missing support for arduino uno r4

itslemcy, 2023-07-09:
#error "Please edit Boards.h with a hardware abstraction for this board"

pgrawehr, 2023-07-09:
To fix this, I need more info. Can you paste the full output of the failed build here? Be sure to set the output to verbose first.

itslemcy, 2023-07-09:
FQBN: arduino:renesas_uno:minima
Using board 'minima' from platform in folder: /Users/username/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.0.1
Using core 'arduino' from platform in folder: /Users/username/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.0.1

Detecting libraries used...
/Users/username/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -w -Os -g3 -fno-use-cxa-atexit -fno-rtti -fno-exceptions -nostdlib -DF_CPU=48000000 -DARDUINO_UNOR4_MINIMA -std=gnu++17 -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsigned-char -ffunction-sections -fdata-sections -fmessage-length=0 -fno-builtin -w -x c++ -E -CC -DARDUINO=10607 "-DPROJECT_NAME=\"/private/var/folders/0_/kcmh2pts77b9z6z3qpw7nh4m0000gn/T/arduino/sketches/842A277FE2625B129A208E4530293BE2/StandardFirmata.ino\"" -DARDUINO_MINIMA -DARDUINO_ARCH_RENESAS_UNO -DARDUINO_ARCH_RENESAS -DARDUINO_FSP -D_XOPEN_SOURCE -mthumb @/Users/username/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.0.1/variants/MINIMA/defines.txt -DCFG_TUSB_MCU=OPT_MCU_RAXXX -I/Users/username/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.0.1/cores/arduino/tinyusb -I/Users/username/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.0.1/cores/arduino/api/deprecated -I/Users/username/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.0.1/cores/arduino/api/deprecated-avr-comp -I/Users/username/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.0.1/cores/arduino -I/Users/username/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.0.1/variants/MINIMA -iprefix/Users/username/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.0.1 @/Users/username/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.0.1/variants/MINIMA/includes.txt /private/var/folders/0_/kcmh2pts77b9z6z3qpw7nh4m0000gn/T/arduino/sketches/842A277FE2625B129A208E4530293BE2/sketch/StandardFirmata.ino.cpp -o /dev/null
Alternatives for Servo.h: [Servo@1.1.8]
ResolveLibrary(Servo.h)
  -> candidates: [Servo@1.1.8]
/Users/username/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -w -Os -g3 -fno-use-cxa-atexit -fno-rtti -fno-exceptions -nostdlib -DF_CPU=48000000 -DARDUINO_UNOR4_MINIMA -std=gnu++17 -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsigned-char -ffunction-sections -fdata-sections -fmessage-length=0 -fno-builtin -w -x c++ -E -CC -DARDUINO=10607 "-DPROJECT_NAME=\"/private/var/folders/0_/kcmh2pts77b9z6z3qpw7nh4m0000gn/T/arduino/sketches/842A277FE2625B129A208E4530293BE2/StandardFirmata.ino\"" -DARDUINO_MINIMA -DARDUINO_ARCH_RENESAS_UNO -DARDUINO_ARCH_RENESAS -DARDUINO_FSP -D_XOPEN_SOURCE -mthumb @/Users/username/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.0.1/variants/MINIMA/defines.txt -DCFG_TUSB_MCU=OPT_MCU_RAXXX -I/Users/username/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.0.1/cores/arduino/tinyusb -I/Users/username/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.0.1/cores/arduino/api/deprecated -I/Users/username/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.0.1/cores/arduino/api/deprecated-avr-comp -I/Users/username/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.0.1/cores/arduino -I/Users/username/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.0.1/variants/MINIMA -I/Users/username/Library/Arduino15/libraries/Servo/src -iprefix/Users/username/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.0.1 @/Users/username/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.0.1/variants/MINIMA/includes.txt /private/var/folders/0_/kcmh2pts77b9z6z3qpw7nh4m0000gn/T/arduino/sketches/842A277FE2625B129A208E4530293BE2/sketch/StandardFirmata.ino.cpp -o /dev/null
Alternatives for Wire.h: [Wire]
ResolveLibrary(Wire.h)
  -> candidates: [Wire]
/Users/username/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -w -Os -g3 -fno-use-cxa-atexit -fno-rtti -fno-exceptions -nostdlib -DF_CPU=48000000 -DARDUINO_UNOR4_MINIMA -std=gnu++17 -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsigned-char -ffunction-sections -fdata-sections -fmessage-length=0 -fno-builtin -w -x c++ -E -CC -DARDUINO=10607 "-DPROJECT_NAME=\"/private/var/folders/0_/kcmh2pts77b9z6z3qpw7nh4m0000gn/T/arduino/sketches/842A277FE2625B129A208E4530293BE2/StandardFirmata.ino\"" -DARDUINO_MINIMA -DARDUINO_ARCH_RENESAS_UNO -DARDUINO_ARCH_RENESAS -DARDUINO_FSP -D_XOPEN_SOURCE -mthumb @/Users/username/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.0.1/variants/MINIMA/defines.txt -DCFG_TUSB_MCU=OPT_MCU_RAXXX -I/Users/username/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.0.1/cores/arduino/tinyusb -I/Users/username/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.0.1/cores/arduino/api/deprecated -I/Users/username/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.0.1/cores/arduino/api/deprecated-avr-comp -I/Users/username/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.0.1/cores/arduino -I/Users/username/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.0.1/variants/MINIMA -I/Users/username/Library/Arduino15/libraries/Servo/src -I/Users/username/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.0.1/libraries/Wire -iprefix/Users/username/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.0.1 @/Users/username/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.0.1/variants/MINIMA/includes.txt /private/var/fol [... cut ...]
/Users/username/Documents/Arduino/libraries/Firmata/Boards.h:1106:2: error: #error "Please edit Boards.h with a hardware abstraction for this board"
 #error "Please edit Boards.h with a hardware abstraction for this board"
  ^~~~~

Using library Servo at version 1.1.8 in folder: /Users/username/Library/Arduino15/libraries/Servo 
Using library Wire in folder: /Users/username/Library/Arduino15/packages/arduino/hardware/renesas_uno/1.0.1/libraries/Wire (legacy)
Using library Firmata at version 2.5.9 in folder: /Users/username/Documents/Arduino/libraries/Firmata 
exit status 1

Compilation error: exit status 1

pgrawehr, 2023-07-10:
Ok, I can give it a try, but I'm not sure it will "just work", because the R4 is technically very different from the R3. 

I'll ping you when I've got something to test, but it may not be tomorrow.

pgrawehr, 2023-07-10:
Also, I'm going to transfer this issue to ConfigurableFirmata, as this supports additional features for newer boards.