firmata/arduino #511: [Question] How to add support for Jolly-dev to StandardFirmataWiFi

ryrych, 2023-10-02:
Hello,
I've purchased Jolly-dev to add support for WiFi to Uno rev3. My goal was to control UNO via Johnny-Five. Unfortunately, UNO with Jolly-dev on board is not supported by the Firmata. Is there any guide how to extend StandardFirmataWiFi to support other boards?

pgrawehr, 2023-10-02:
Wifi is supported on ConfigurableFirmata,  and it once was running with Uno as well. I have not tested that recently since I don't have a Uno WiFi board, but it should clearly be possible to get that to work again.

What features of joly-dev do you need in particular?

ryrych, 2023-10-04:
@pgrawehr Thank you, I'll try this out.

>  What features of joly-dev do you need in particular?
Just WiFi: being able to tether via WiFi

pgrawehr, 2023-10-04:
Oh, got it. I was mislead by the name and thought you where talking about another device. 

Unfortunately, the documentation of this joly-dev module is very poor, and I can't find information about how you would use its wifi capability in general. Where you able to build a simple example using the WIFI interface? (e.g. just a remotely-commanded LED)?

ryrych, 2023-10-04:
Not yet, I'll test it out till the end of the week.

ryrych, 2023-10-04:
@pgrawehr Hi,
I was able to control LED via wifi:

[image]

with example: Jolly > WiFi > WiFiRestServer.ino

ryrych, 2023-10-05:
When I compile `ConfigurableFirmata` I get this error:

```
Arduino/libraries/ConfigurableFirmata/src/utility/Boards.h:828:2: error: #error "Please edit Boards.h with a hardware abstraction for this board"
 #error "Please edit Boards.h with a hardware abstraction for this board"
  ^~~~~

exit status 1

Compilation error: exit status 1
```

pgrawehr, 2023-10-05:
Ok, that's kind of expected. You need to add your hardware to that board. Since it is electrically equivalent to an Arduino Uno, you can just add your board to that case. 

Edit line 145 of Boards.h, it currently reads

```
#if defined(__AVR_ATmega168__) || defined(__AVR_ATmega328P__) || defined(__AVR_ATmega328__)
```

You need to add the name of the board type. The easiest way is to look at the output of the build. The command lines show a set of -D.... - arguments, one of which will probably say something like AVR_JOY_DEV or so.

ryrych, 2023-10-07:
Hi,
I changed the line to:

```
#if defined(__AVR_ATmega168__) || defined(__AVR_ATmega328P__) || defined(__AVR_ATmega328__) || defined(ARDUINO_jolly)
```

`ARDUINO_jolly` is taken from compilling `WiFiRestServer.ino`: `-DARDUINO=10607 -DARDUINO_jolly -DARDUINO_ARCH_AVR -Djolly`

When `#define ENABLE_WIFI` is commented out, I can compile the `ConfigurableFirmata`, but when I enable WiFi, I get lots of errors:

```
FQBN: jolly:avr:jolly:cpu=esp
Using board 'jolly' from platform in folder: Users/user/Library/Arduino15/packages/jolly/hardware/avr/1.0.1
Using core 'atmega328pb' from platform in folder: Users/user/Library/Arduino15/packages/jolly/hardware/avr/1.0.1

…

Compiling sketch...
Users/user/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega328pb -DESP_CH_SPI -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_jolly -DARDUINO_ARCH_AVR -Djolly -IUsers/user/Library/Arduino15/packages/jolly/hardware/avr/1.0.1/cores/atmega328pb -IUsers/user/Library/Arduino15/packages/jolly/hardware/avr/1.0.1/variants/mega328pb -IUsers/user/Documents/Arduino/libraries/ConfigurableFirmata/src -IUsers/user/Documents/Arduino/libraries/WiFiNINA/src -IUsers/user/Library/Arduino15/packages/jolly/hardware/avr/1.0.1/libraries/Wire/src -IUsers/user/Library/Arduino15/packages/jolly/hardware/avr/1.0.1/libraries/SPI/src -IUsers/user/Library/Arduino15/packages/jolly/hardware/avr/1.0.1/libraries/SoftwareSerial/src -IUsers/user/Documents/Arduino/libraries/DHT_sensor_library -IUsers/user/Documents/Arduino/libraries/Adafruit_Unified_Sensor /private/var/folders/bd/rswv6vb17wq86p4zq2z5plm40000gn/T/arduino/sketches/C2F58CB8566BF84B3BC1919F60E40748/sketch/ConfigurableFirmata.ino.cpp -o /private/var/folders/bd/rswv6vb17wq86p4zq2z5plm40000gn/T/arduino/sketches/C2F58CB8566BF84B3BC1919F60E40748/sketch/ConfigurableFirmata.ino.cpp.o
In file included from /private/var/folders/bd/rswv6vb17wq86p4zq2z5plm40000gn/T/.arduinoIDE-unsaved202397-75619-10v1vyg.tgp9/ConfigurableFirmata/ConfigurableFirmata.ino:5:0:
Users/user/Documents/Arduino/libraries/ConfigurableFirmata/src/ConfigurableFirmata.h:56:33: error: expected identifier before numeric constant
 #define SET_PIN_MODE            0xF4 // set a pin to INPUT/OUTPUT/PWM/etc
                                 ^
Users/user/Documents/Arduino/libraries/WiFiNINA/src/utility/wifi_spi.h:102:5: note: in expansion of macro 'SET_PIN_MODE'
     SET_PIN_MODE  = 0x50,
     ^~~~~~~~~~~~
Users/user/Documents/Arduino/libraries/ConfigurableFirmata/src/ConfigurableFirmata.h:56:33: error: expected '}' before numeric constant
 #define SET_PIN_MODE            0xF4 // set a pin to INPUT/OUTPUT/PWM/etc
                                 ^
Users/user/Documents/Arduino/libraries/WiFiNINA/src/utility/wifi_spi.h:102:5: note: in expansion of macro 'SET_PIN_MODE'
     SET_PIN_MODE  = 0x50,
     ^~~~~~~~~~~~
Users/user/Documents/Arduino/libraries/ConfigurableFirmata/src/ConfigurableFirmata.h:56:33: error: expected unqualified-id before numeric constant
 #define SET_PIN_MODE            0xF4 // set a pin to INPUT/OUTPUT/PWM/etc
                                 ^
Users/user/Documents/Arduino/libraries/WiFiNINA/src/utility/wifi_spi.h:102:5: note: in expansion of macro 'SET_PIN_MODE'
     SET_PIN_MODE  = 0x50,
     ^~~~~~~~~~~~
In file included from Users/user/Documents/Arduino/libraries/WiFiNINA/src/utility/wifi_drv.h:25:0,
                 from Users/user/Documents/Arduino/libraries/WiFiNINA/src/WiFiStorage.h:23,
                 from Users/user/Documents/Arduino/libraries/WiFiNINA/src/WiFi.h:38,
                 from Users/user/Documents/Arduino/libraries/WiFiNINA/src/WiFiNINA.h:23,
                 from /private/var/folders/bd/rswv6vb17wq86p4zq2z5plm40000gn/T/.arduinoIDE-unsaved202397-75619-10v1vyg.tgp9/ConfigurableFirmata/ConfigurableFirmata.ino:65:
Users/user/Documents/Arduino/libraries/WiFiNINA/src/utility/wifi_spi.h:117:1: error: expected declaration before '}' token
 };
 ^

Using library ConfigurableFirmata at version 3.2.0 in folder: Users/user/Documents/Arduino/libraries/ConfigurableFirmata 
Using library WiFiNINA at version 1.8.14 in folder: Users/user/Documents/Arduino/libraries/WiFiNINA 
Using library Wire at version 1.0 in folder: Users/user/Library/Arduino15/packages/jolly/hardware/avr/1.0.1/libraries/Wire 
Using library SPI at version 1.0 in folder: Users/user/Library/Arduino15/packages/jolly/hardware/avr/1.0.1/libraries/SPI 
Using library SoftwareSerial at version 1.0 in folder: Users/user/Library/Arduino15/packages/jolly/hardware/avr/1.0.1/libraries/SoftwareSerial 
Using library DHT sensor library at version 1.4.4 in folder: Users/user/Documents/Arduino/libraries/DHT_sensor_library 
Using library Adafruit Unified Sensor at version 1.1.12 in folder: Users/user/Documents/Arduino/libraries/Adafruit_Unified_Sensor 
exit status 1

Compilation error: exit status 1
```

In line `:64` I tried replacing `<WiFi.h>` with `<WiFiNINA.h>`:

```
#ifdef ENABLE_WIFI
// #include <WiFi.h>
#include <WiFiNINA.h>
```

as `<WiFi.h>` generates

```
/ConfigurableFirmata/ConfigurableFirmata.ino: In function 'void initTransport()':
/ConfigurableFirmata/ConfigurableFirmata.ino:158:7: error: 'class WiFiClass' has no member named 'mode'
  WiFi.mode(WIFI_STA);
       ^~~~
/ConfigurableFirmata/ConfigurableFirmata.ino:158:12: error: 'WIFI_STA' was not declared in this scope
  WiFi.mode(WIFI_STA);
            ^~~~~~~~
/ConfigurableFirmata/ConfigurableFirmata.ino:158:12: note: suggested alternative: 'WIFI_STREAM_H'
  WiFi.mode(WIFI_STA);
            ^~~~~~~~
            WIFI_STREAM_H
Multiple libraries were found for "WiFi.h"
  Used: /Users/User/Library/Arduino15/packages/jolly/hardware/avr/1.0.1/libraries/WiFi
  Not used: /Users/User/Documents/Arduino/libraries/WiFiNINA
```

pgrawehr, 2023-10-08:
@ryrych It appears that Wifi.h is providing some conflicting definitions. Try to add `#undef SET_PIN_MODE` before the first line that throws an error.