firmata/arduino #514: Problem compiling firmata for Giga R1 Wifi board

jmarsh12, 2024-01-31:
I'm trying to use a firmata example for bluetooth which is supposedly compatible with the R1 Wifi but I'm hit with this error. I know this has been resolved with other boards in the past by adding the abstraction for the board in Boards.h. I know the Giga footprint is similar to the Mega, but I'm unsure what would need to be put in the abstraction for this board. The error is listed below:

Generating function prototypes...
C:\Users\JAM\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4/bin/arm-none-eabi-g++ -c -w -g3 -nostdlib @C:\Users\JAM\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.10\variants\GIGA/defines.txt @C:\Users\JAM\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.10\variants\GIGA/cxxflags.txt -mcpu=cortex-m7 -mfloat-abi=softfp -mfpu=fpv5-d16 -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_GIGA -DARDUINO_ARCH_MBED_GIGA -DARDUINO_ARCH_MBED -DARDUINO_LIBRARY_DISCOVERY_PHASE=1 -IC:\Users\JAM\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.10\cores\arduino -IC:\Users\JAM\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.10\variants\GIGA -IC:\Users\JAM\AppData\Local\Arduino15\libraries\Servo\src -IC:\Users\JAM\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.10\libraries\Wire -IC:\Users\JAM\Documents\Arduino\libraries\Firmata -DCM4_BINARY_START=0x60000000 -DCM4_BINARY_END=0x60040000 -DCM4_RAM_END=0x60080000 -IC:\Users\JAM\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.10\cores\arduino/api/deprecated -IC:\Users\JAM\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.10\cores\arduino/api/deprecated-avr-comp -iprefixC:\Users\JAM\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.10\cores\arduino @C:\Users\JAM\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.10\variants\GIGA/../GIGA/includes.txt C:\Users\JAM\AppData\Local\Temp\arduino\sketches\AE848B14A42C67E61CB81661068E5993\sketch\StandardFirmataBLE.ino.cpp -o C:\Users\JAM\AppData\Local\Temp\2950846757\sketch_merged.cpp
In file included from C:\Users\JAM\Documents\Arduino\libraries\Firmata/Firmata.h:17:0,
                 from C:\Users\JAM\AppData\Local\Temp\.arduinoIDE-unsaved2024031-9832-16m7plk.l22k\StandardFirmataBLE\StandardFirmataBLE.ino:28:
C:\Users\JAM\Documents\Arduino\libraries\Firmata/Boards.h:1106:2: error: #error "Please edit Boards.h with a hardware abstraction for this board"
 #error "Please edit Boards.h with a hardware abstraction for this board"
  ^~~~~
Multiple libraries were found for "Firmata.h"
  Used: C:\Users\JAM\Documents\Arduino\libraries\Firmata
  Not used: C:\Users\JAM\AppData\Local\Arduino15\libraries\Firmata
Using library Servo at version 1.2.1 in folder: C:\Users\JAM\AppData\Local\Arduino15\libraries\Servo 
Using library Wire in folder: C:\Users\JAM\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.10\libraries\Wire (legacy)
Using library Firmata at version 2.5.9 in folder: C:\Users\JAM\Documents\Arduino\libraries\Firmata 
exit status 1

Compilation error: exit status 1

pgrawehr, 2024-01-31:
Can you give a link to the board description from the vendor? That will usually have the necessary information.

jmarsh12, 2024-01-31:
> Can you give a link to the board description from the vendor? That will usually have the necessary information.

Added a link to the technical specs and datasheet. Thanks for the reminder!