firmata/arduino #518: Firmata protocol as I2C Slave

EMonatko, 2024-12-17:
Hi,

Does the protocol support I2C slave mode?
Can I configure my Arduino as a slave using this Firmata?

Thanks in advance!

pgrawehr, 2024-12-19:
No, that is currently not possible. Typically an Arduino using Firmata is used to control sensors from a PC, so this seems to be a strange requirement. What is your use case?

EMonatko, 2024-12-19:
I'm my case i need to receive from specific sensor data just to check if my i2c cables are working correctly

pgrawehr, 2024-12-19:
There should be easier ways to achieve this. The best solution of course would be to use an oscilloscope. And you can start out by doing the standard I2C bus scan (available as an example in Arduino IDE). That way you can see whether the sensor responds.

PizzaProgram, 2025-04-14:
> And you can start out by doing the standard I2C bus scan

I think doing so is **interrupting** I2C communication with devices currently working, so it's very dangerous!
Because the "scanning" is based on: WRITING to each address to test each. See:
https://learn.adafruit.com/scanning-i2c-addresses/arduino

pgrawehr, 2025-04-15:
@PizzaProgram Sure, but if you just want to test whether the cabling is ok, that's the way to go. Obviously, communication has not been established (and thus the devices aren't initialized) when you start up and first want to verify whether the hardware is operational and connected.